ARMmbed/mbed-os #5133: NRF51_DK - AnalogIn, PWM, Concurrent GPIO failures (CI test shield)

scartmell-arm, 2017-09-18:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF51_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
6

**mbed-cli version:**
1.2.0

**mbed-os sha:**
Current master `https://github.com/ARMmbed/mbed-os/#0906be6397c984026df180b07a3bbd965af52346`

**Expected behavior**
All tests should pass.

**Actual behavior**
```
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target           | platform_name | test suite                  | test case                                                              | passed | failed | result | elapsed_time (sec) |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_0                                                  | 0      | 2      | FAIL   | 0.27               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_1                                                  | 0      | 2      | FAIL   | 0.27               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_2                                                  | 0      | 2      | FAIL   | 0.28               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_3                                                  | 0      | 2      | FAIL   | 0.28               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_4                                                  | 0      | 2      | FAIL   | 0.27               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-analogin          | Analog Input on AIN_5                                                  | 0      | 2      | FAIL   | 0.27               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_fall          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise_fall     | PWM_0 Frequency 10ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-spi               | SPI - SD card exists                                                   | 0      | 1      | FAIL   | 2.33               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-comms      | Concurrent testing of I2C and SPI in a single thread                   | 0      | 3      | FAIL   | 5.61               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-comms      | Concurrent testing of I2C and SPI in multiple threads                  | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D2,D3), AnalogIn(A0), and InterruptIn(D4,D5) | 0      | 2      | FAIL   | 0.19               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D3,D2), AnalogIn(A1), and InterruptIn(D5,D4) | 0      | 2      | FAIL   | 0.2                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D4,D5), AnalogIn(A2), and InterruptIn(D6,D7) | 0      | 2      | FAIL   | 0.19               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D5,D4), AnalogIn(A3), and InterruptIn(D7,D6) | 0      | 12     | FAIL   | 0.79               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D6,D7), AnalogIn(A4), and InterruptIn(D8,D9) | 0      | 12     | FAIL   | 0.78               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D7,D6), AnalogIn(A5), and InterruptIn(D9,D8) | 0      | 12     | FAIL   | 0.78               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D8,D9), AnalogIn(A5), and InterruptIn(D2,D3) | 0      | 2      | FAIL   | 0.2                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-gpio       | Concurrent testing of DIO(D9,D8), AnalogIn(A5), and InterruptIn(D3,D2) | 0      | 12     | FAIL   | 0.79               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in a single thread                | 0      | 3      | FAIL   | 19.25              |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in multiple threads               | 0      | 0      | ERROR  | 0.0                |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: completed in 1020.51 sec
mbedgt: exited with code 8
```

**Steps to reproduce**

Run the application at https://github.com/ARMmbed/ci-test-shield

i.e.
```
$ git clone https://github.com/ARMmbed/ci-test-shield.git && cd ci-test-shield
```

Set mbed-os version in `mbed-os.lib`

```
$ mbed deploy
$ mbed target auto
$ mbed toolchain GCC_ARM
$ mbed test -n tests-* --app-config .\mbed_app.json
```

pan-, 2017-09-19:
The NRF51 DK is not Arduino compatible out of the box, it requires some hardware rework as indicated in the [user guide](https://www.nordicsemi.com/eng/layout/set/print/nordic/download_resource/38677/1/88887966/64339). 
Unfortunately, there's no way to completely disable Buttons (mapped from D5 to D9). That might explains some of the issues you are experiencing.

0xc0170, 2017-09-19:
> Unfortunately, there's no way to completely disable Buttons (mapped from D5 to D9). That might explains some of the issues you are experiencing.

@nvlsianpu @anangl Can you confirm that the issues above are known and expected?

nvlsianpu, 2017-09-26:
Analog part was failing because input voltage range is set to 0-1.2 V, which causes input saturation. I will extend this to 3.6 V using build in analog prescaler.

pan-, 2017-09-26:
@nvlsianpu Could you comment about the buttons ?

nvlsianpu, 2017-09-26:
I'm investigating this.

nvlsianpu, 2017-09-27:
This is all software related bugs - two patches for fix them are pending.

scartmell-arm, 2017-10-10:
Thanks, I ran this again with the latest mbed-os and got the following results

```
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target           | platform_name | test suite                  | test case                                                              | passed | failed | result | elapsed_time (sec) |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_fall          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise_fall     | PWM_0 Frequency 10ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-comms      | Concurrent testing of I2C and SPI in a single thread                   | 0      | 3      | FAIL   | 5.61               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-comms      | Concurrent testing of I2C and SPI in multiple threads                  | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in a single thread                | 0      | 3      | FAIL   | 19.28              |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in multiple threads               | 0      | 0      | ERROR  | 0.0                |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+

```

pan-, 2017-10-10:
@ARMmbed/team-nordic Could you look at the remaining issues ?

frkv, 2017-10-10:
The Nordic SDK team is very busy with an imminent SDK release. A lot of extra activities have been put on hold, as we are going out with a whole new architecture of our chips. We will look at this as soon as we can, but it may take some time.

nvlsianpu, 2017-10-10:
@scartmell-arm And what were results for NRF52_DK? Didn't they pass?

nvlsianpu, 2017-10-10:
looks like after my last update to SPI was on rev c2951877509fcdd8b8152bf78379a14d6f4726c9. For this spi tests had passed .After this some changes corrupted spi - still not sure which one.

nvlsianpu, 2017-10-10:
tests-api-spi failing since rev cb4e9b32a2933553630ff1ed40f9bdb77e51fda2 Sleep HAL: add sleep manager API

scartmell-arm, 2017-10-12:
> @scartmell-arm And what were results for NRF52_DK? Didn't they pass?

I didn't have one when I did the testing, but have since located one. I reran the test for both boards. (I'm not sure why the results are different for NRF51 this time).

NRF51_DK
```
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target           | platform_name | test suite                  | test case                                                              | passed | failed | result | elapsed_time (sec) |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_fall          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise          | PWM_0 Frequency 30ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-pwm_rise_fall     | PWM_0 Frequency 10ms                                                   | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-comms      | Concurrent testing of I2C and SPI in multiple threads                  | 0      | 0      | ERROR  | 0.0                |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in multiple threads               | 0      | 0      | ERROR  | 0.0                |
+------------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
```
NRF52_DK
```
+------------------+---------------+-----------------------------+-----------------------------+--------+--------+--------+--------------------+
| target           | platform_name | test suite                  | test case                   | passed | failed | result | elapsed_time (sec) |
+------------------+---------------+-----------------------------+-----------------------------+--------+--------+--------+--------------------+
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-analogin          | tests-api-analogin          | 0      | 1      | ERROR  | 125.82             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-businout          | tests-api-businout          | 0      | 1      | ERROR  | 125.77             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-digitalio         | tests-api-digitalio         | 0      | 1      | ERROR  | 125.76             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-i2c               | tests-api-i2c               | 0      | 1      | ERROR  | 125.8              |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-interruptin       | tests-api-interruptin       | 0      | 1      | ERROR  | 125.9              |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-pwm_fall          | tests-api-pwm_fall          | 0      | 1      | ERROR  | 125.94             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-pwm_rise          | tests-api-pwm_rise          | 0      | 1      | ERROR  | 125.87             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-pwm_rise_fall     | tests-api-pwm_rise_fall     | 0      | 1      | ERROR  | 125.83             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-api-spi               | tests-api-spi               | 0      | 1      | ERROR  | 126.03             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-analogin  | tests-assumptions-analogin  | 0      | 1      | ERROR  | 125.87             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-digitalio | tests-assumptions-digitalio | 0      | 1      | ERROR  | 126.1              |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-i2c       | tests-assumptions-i2c       | 0      | 1      | ERROR  | 125.73             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-pwm       | tests-assumptions-pwm       | 0      | 1      | ERROR  | 125.9              |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-pwmout    | tests-assumptions-pwmout    | 0      | 1      | ERROR  | 107.13             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-assumptions-spi       | tests-assumptions-spi       | 0      | 1      | ERROR  | 19.09              |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-concurrent-comms      | tests-concurrent-comms      | 0      | 1      | ERROR  | 125.94             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-concurrent-gpio       | tests-concurrent-gpio       | 0      | 1      | ERROR  | 125.76             |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-concurrent-mixed      | tests-concurrent-mixed      | 0      | 1      | ERROR  | 126.1              |
+------------------+---------------+-----------------------------+-----------------------------+--------+--------+--------+--------------------+
```

scartmell-arm, 2017-10-12:
I guess there's either something wrong or I haven't configured my NRF52 properly as I get the same timeout errors both before and after the changeset highlighted in [nRF52 SPI had been broken by `Add sleep manager API` PR #5297 ](https://github.com/ARMmbed/mbed-os/issues/5297). I'll look into it.

nvlsianpu, 2017-11-17:
This is what I have on terminal for 'concurrent comms' test for NRF52_DK:

```
mbedmbedmbedmbedmbedmbedmbedmbed
                                       
{{__sync;1a2cdb58-88d8-4c5f-8db4-0d4609576611}}
                               
{{__version;1.3.0}}
                                                           
{{__timeout;40}}
                                                              
{{__host_test_name;default_auto}}
                                             
>>> Running 2 test cases...
                                                    
{{__testcase_name;Concurrent testing of I2C and SPI in a single thread}}
      
{{__testcase_name;Concurrent testing of I2C and SPI in multiple threads}}
     

                                                                               
>>> Running case #1: 'Concurrent testing of I2C and SPI in a single thread'...
 
{{__testcase_start;Concurrent testing of I2C and SPI in a single thread}}
     
{{__testcase_finish;Concurrent testing of I2C and SPI in a single thread;1;0}}

>>> 'Concurrent testing of I2C and SPI in a single thread': 1 passed, 0 failed
 

                                                                               
>>> Running case #2: 'Concurrent testing of I2C and SPI in multiple threads'...

{{__testcase_start;Concurrent testing of I2C and SPI in multiple threads}}
    
Operator new[] out of memory
```

Looks like it is a stack to small issue. @pan- your opinion?

nvlsianpu, 2017-11-27:
@pan-, @0xc0170 ^^

0xc0170, 2017-11-27:
@nvlsianpu Seems you are correct, looking at the code, they use `OS_STACK_SIZE` for the app, that is very small, and should not be used. Can you send a patch to address this issue?

linlingao, 2019-05-29:
This issue is 18month old. Let me know if it's still an issue.

closed 2019-05-29 by linlingao