ARMmbed/mbed-os #5134: Binary built with export not working - Nordic nRF52-DK

deepikabhavnani, 2017-09-18:
Mbed-os-example-ble-BatteryLevel application for the Nordic nRF52-DK board if built with the web based mbed toolchain, works properly when downloaded the resulting .hex file to the board (JLINK device), but fails when exported and built with EWARM. It appears that it gets linked to the wrong address and doesn't start.

- Type: Bug
- Priority: Major

## Bug

**Target**
Nordic nRF52-DK

**Toolchain:**
IIAR

**mbed-os version**
**Steps to reproduce**

deepikabhavnani, 2017-09-18:
@TTornblom - I have added issue, export to IAR should work. Please add "mbed-os version" and "Steps to reproduce" detail.

TTornblom, 2017-09-19:
mbed-os 5.5.7 RC

Steps to reproduce, using web based mbed:

1) Add Nordic nRF52-DK board to Registered Platforms

2) Create new program for that platform using "[mbed OS 5] BLE example for Battery Level Service"

3) Set windows environment variable MBED_IAR_PATH to point to EW 7.80 root (C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5). This example does not work with EW8 due to references to FILE *, which is not available in EW8.

4) Select "mbed-os-example-ble-BatteryLevel" program in Program Workspace list

5) Export program using IAR Export Toolchain

6) Unpack downloaded zip archive

7) In EW 7.80, open the just unpacked work space

8) Follow the Known issues paragraph on [https://os.mbed.com/platforms/Nordic-nRF52-DK/](url) to make sure the board has the correct firmware and that the Debugger category is setup as recommended.

9) Open project options Output Converter, notice that the output format is Motorola, which is not what the mbed toolchain uses. I believe the correct option here is Intel extended

10) Build program

11) Download the output file to the JLINK disk device

 The program does not start, which is indicated by LED1 not blinking as it does with the file built by the mbed toolchain.

Comparison of the first few lines of the output files, which shows that the addresses are different:

mbed:
```
:020000040000FA
:1000000000040020E508000079050000C508000094
:10001000830500008D05000097050000000000002A
:1000200000000000000000000000000009090000BE
```

EW 7.80:
```
:020000040001F9
:10C000001055002021C0020039C002003BC00200D0
:10C010003DC002003FC0020041C00200000000001D
:10C020000000000000000000000000009DB90200B8
```

0xc0170, 2017-09-19:
cc @pan- @nvlsianpu @anangl

TTornblom, 2017-09-22:
I have now tested a few other examples for this board, BLE_Button and BLE_LED, for mbed OS 2, and they suffer the same issues: produces Motorola srec files, apparently linked to the wrong address, and doesn't work.

I assume they all suffer from the same export to IAR issues.

theotherjimmy, 2017-09-22:
> Download the output file to the JLINK disk device

This step is probably what's causing the problems. Nordic devices use a soft device, and the soft device is assumed to be present on the device. A user of the IDE exporters must flash from the IDE, as the generated hex/bin/axf (just elf, incorrectly named) do not have this soft device and therefore will not work.

TTornblom, 2017-09-25:
So this device is included in the hex file produced by the mbed toolchain, but not when exporting the project?

Is this by design, or is it just an omission to export the required soft device to the IDE?

theotherjimmy, 2017-09-25:
@TTornblom This is by design. We expect the exporter to write over only the application's location in flash. 

That being said, I have no problem with changing that behavior, and you're the expert on this IDE.

TTornblom, 2017-09-29:
We're a bit confused about this behavior.

Is there any advantage with the current functionality over having it behave the same as when building with the mbed toolchain?

Does it work the same for other vendors/boards?

Personally I find it would make sense if they worked the same for at least the following reasons:

1) Eliminate confusion
2) Avoid having to use external debug hardware, as long as it **can** be used if needed
3) Make it easier to compare the various toolchains

theotherjimmy, 2017-09-29:
@TTornblom 
>We're a bit confused about this behavior.

The exporters are designed to mimic the behavior of `mbed compile` exactly, except where it would break the expectations of an user of the IDE that the exporter targets.
At this point this may be legacy behavior that we can change. More on that below.

> [have] it behave the same as when building with [`mbed compile`]

I agree that would be nice, and a better design, if it's not breaking any expectations that a user of the IAR IDE would have. If there is a switch, or series of settings, that "Does the right thing", I would like to turn that switch on. Doing the right thing in this context is two things: the debug button in the IDE still works, and the IDE builder produces a hex file that can be copied to the MSD. I was unable to find such a such a switch when I last looked. I'm very happy to be wrong about the absence of one and would approve a PR to enable that switch for Nordic devices in a heartbeat.

An aside: You seem to be referring to *the mbed toolchain*. I think what you're referring to is the behavior of the `mbed compile` subcommand in this context, which we usually call mbed compile.

TTornblom, 2017-10-10:
Obviously I had misunderstood how this board works. I didn't realize it has an on-board J-Link, which works great with IAR Embedded Workbench. The example I exported from mbed works great when run under C-SPY, and I see no real reason to change anything.

Sorry for the noise.

0xc0170, 2017-10-13:
> Sorry for the noise.

Does this mean this issue can be closed as resolved?

closed 2017-11-21 by deepikabhavnani