ARMmbed/mbed-os #5135: EVK-ODIN-W2: 'AnalogOut' does not name a type

andrewc-arm, 2017-09-19:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
EVK-ODIN-W2

**Toolchain:**
GCC_ARM

----------------------------------------------------------------

Hi, I used following code to compile K64F using AnalogOut.
```
#include "mbed.h"

AnalogOut  aout5(A5);

int main() {
	int aoutD = 0;
	while (true) {
		aoutD = (aoutD + 1) % 10;
		aout5 = (float)(aoutD) * 0.1f;
        wait(0.5);
    }
}
```

Do you know why I cannot build it from EVK-ODIN-W2?

Thanks,

andrewc-arm, 2017-09-19:
I see that the board does not support AnalogOut as @toyowata mentioned in a separate email.
```
Hi Andrew,

The target inherits “FAMILY_STM32” here: 
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1687

And the “FAMILY_STM32” does not support “ANALOGOUT” feature here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L677

Also, pinout of the target does not mention about ANALOGOUT:
https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/#pin-layout

So, I think the target does not have AnalogOut feature.

Cheers,
Toyo
```

closed 2017-09-19 by andrewc-arm

jeromecoutant, 2017-09-19:
Hi

Just to comment Toyowata mail:

UBLOX_EVK_ODIN_W2 MCU is STM32F439ZI.
All STM32 doesn't support ANALOGOUT, that's why "FAMILY_STM32" doesn't include it.
STM32F439ZI is supporting AnalogOut, so it could be added in the "device_has_add" line from targets.json file. The only issue is the available pins in the board.

Jerome

0xc0170, 2017-09-19:
> STM32F439ZI is supporting AnalogOut, so it could be added in the "device_has_add" line from targets.json file. The only issue is the available pins in the board.

@andreaslarssonublox @andreaspeterssonublox Is that correct, no analog out for odin target?

andreaslarssonublox, 2017-09-19:
@0xc0170  Yes, that's correct. The ODIN-W2 module does not have any of the analog out capable pins available from the STM32F439ZI.