ARMmbed/mbed-os #5136: LowPowerTimeout 500us test not within specification for NUCLEO_F746ZG

studavekar, 2017-09-19:
### Description

LowPowerTimeout 500us test not within specification for NUCLEO_F746ZG 

https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_drivers/lp_timeout/main.cpp

```
[1505797061.50][CONN][RXD] >>> Running case #1: '500us LowPowerTimeout'...
[1505797061.54][CONN][INF] found KV pair in stream: {{__testcase_start;500us LowPowerTimeout}}, queued...
[1505797061.62][CONN][RXD] :98::FAIL: Values Not Within Delta 600 Expected 500 Was 123536
[1505797061.62][CONN][INF] found KV pair in stream: {{__testcase_finish;500us LowPowerTimeout;0;1}}, queued...

```

[LowPowerTimeout_NUCLEO_F746ZG.txt](https://github.com/ARMmbed/mbed-os/files/1313167/LowPowerTimeout_NUCLEO_F746ZG.txt)

- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F746ZG 

**Toolchain:**

GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
a0d5503 Merge pull request #5116 from hasnainvirk/iar_fix

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

run the below command in loop

```
mbed test -m NUCLEO_F746ZG -t GCC_ARM -n tests-mbed_drivers-lp_timeout -v
```

0xc0170, 2017-09-19:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-09-26:
ST_INTERNAL_REF 38311

jeromecoutant, 2017-09-27:
Hi
Please check #5208

jeromecoutant, 2017-10-05:
Can we close this issue ?
Thx

closed 2017-10-05 by 0xc0170