ARMmbed/mbed-os #5139: OdinWiFiInterface crashes

SeppoTakalo, 2017-09-19:
### Description
- Type: Bug
- Priority: Major
-----------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
`ca661f9d2`

`5.5.7`

**Expected behavior**
Driver initialises itself.

**Actual behavior**
Driver crashes, program does not continue.

**Steps to reproduce**
```
$ mbed new odin-test
$ cd odin-test
```

Edit `main.cpp` to contain following

```
#include "mbed.h"
#include "OdinWiFiInterface.h"

int main()
{
    printf("Initialising OdinWiFiInterface\r\n");
    wait(0.5);
    OdinWiFiInterface wifi;

    printf("Done\r\n");
    return 0;
}
```

Build & flash.

Check serial port:
`Initialising OdinWiFiInterface` is printed but `Done` does not appear.

Works on version `5.5.1` and `5.4.7`

SeppoTakalo, 2017-09-19:
| tag/branch | works |
|--------|-------|
| master | no |
| mbed-os-5.6.0-rc1 | no |
| mbed-os-5.5.7 | no |
| mbed-os-5.5.1 | yes |
| mbed-os-5.5.0 | no |
| mbed-os-5.4.7 | yes |
| mbed-os-5.4.5 | yes |

This seems to have been known problem in Mbed OS 5.5 release time, but never reported
in the known issues list. Fix seems to have been merged in 5.5.1, but re-appears later.

* https://github.com/ARMmbed/mbed-os-example-wifi/issues/36
* https://github.com/ARMmbed/mbed-os/issues/4663
* https://github.com/ARMmbed/mbed-os/pull/4567

0xc0170, 2017-09-19:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-09-29:
I tested this and I could reproduce the same way as above. However, when using the latest mbed-os master(bb61b42fba8a8b3083b5b714a12c11baae2041d3) it seems to work.

0xc0170, 2017-09-29:
> I tested this and I could reproduce the same way as above. However, when using the latest mbed-os master(bb61b42) it seems to work.

Can you test 5.6.0 ? 5.6.1 could also be tested from a 5.6 branch (will be published soon).

andreaslarssonublox, 2017-09-29:
I tested this:
Branch mbed-os-5.6 (cc7556a92fb9320f4bebb190c6e1315af116c50c)
Works

Tag mbed-os-5.6.0-rc3 (5499db1e815b035c2039b6f1e7cbcbf1fd7e0f19)
Does not work

I will continue looking at why it fails.

andreaslarssonublox, 2017-10-02:
I have compiled a new driver based on mbed-os 5.5.6 and that seems to work fine for all versions above. We will most likely release it later this week.

SeppoTakalo, 2017-10-09:
This seems to be fixed now in Mbed OS 5.6.2 and master branch.

Can somebody verify whether fix was applied or do I fail to reproduce the problem?

SeppoTakalo, 2017-10-09:
I tried to bisect and looks like the behaviour changed in https://github.com/ARMmbed/mbed-os/pull/5080

Can this be the corrective action or is this just a side effect, (or me not bisecting correctly).

teetak01, 2017-10-09:
It would seem that already Mbed OS 5.6.1 fixed this issue.

0xc0170, 2017-10-13:
> It would seem that already Mbed OS 5.6.1 fixed this issue.

@SeppoTakalo Can you confirm, that since 5.6.1 this is working?

JanneKiiskila, 2017-10-14:
I don't think this is limited to WiFI only, I think it's also the ethernet stack. I.e. any networking you enable on the Odin will crash like this.

SeppoTakalo, 2017-10-17:
@0xc0170 I did already confirm earlier that 5.6.2 is working.
The root cause, as I bisected, was in HW crypto engine.

JanneKiiskila, 2017-10-17:
For my case it turned out to be a flashing issue in the end. I just got the serial number and nothing else, but the example wasn't printing much.

SeppoTakalo, 2018-03-08:
Re-tested this:

| tag/branch | works |
|--------|-------|
| master(#79bd2638) | yes |
| mbed-os-5.7.6 | no |
| mbed-os-5.7.0 | yes |

SeppoTakalo, 2018-06-12:
I'm inclined to close this as it does not seems to happen anymore with 5.8.x or 5.9 branches.

closed 2018-09-26 by screamerbg