ARMmbed/mbed-os #5142: [OOB 5.6 r2] make_armc6 exporter does not support NRF52_DK and other target

toyowata, 2017-09-20:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK and K64F (and others?)

**Toolchain:**
ARMC6

**Toolchain version:**
Product: MDK Professional 5.24
Component: ARM Compiler 6.7
Tool: armclang [5c9efc00]

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed-os-example-blinky (4230f67abdcb)
`- mbed-os (6e087488e101)

**Expected behavior**
```mbed export -m NRF52_DK -i make_armc6``` create Makefile.

**Actual behavior**
```mbed export -m NRF52_DK -i make_armc6``` create Makefile.

```
E:\oob\mbed-os-example-blinky> mbed export -m NRF52_DK -i make_armc6
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--source SOURCE_DIR]
                  [-D MACROS] [--profile PROFILE] [--update-packs]
                  [--app-config APP_CONFIG]
project.py: error: NRF52_DK not supported by make_armc6
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u E:\oob\mbed-os-example-blinky\mbed-os\tools\project.py -i make_armc6 -m NRF52_DK --source ." in "E:\oob\mbed-os-example-blinky"
---
E:\oob\mbed-os-example-blinky>
```
**Steps to reproduce**

$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.6-oob2
$ mbed export -m NRF52_DK -i make_armc6

0xc0170, 2017-09-20:
cc @theotherjimmy

adbridge, 2017-09-20:
@theotherjimmy I thought we'd fixed the armc6 exporters for rc2 ?

theotherjimmy, 2017-09-20:
@adbridge That's a `not supported` result. I don't think that's really a failure.

theotherjimmy, 2017-09-20:
I disagree with the analysis that this is a bug. If anything, it's an enhancement request.

adbridge, 2017-09-20:
It was the 'other targets' bit that was more concerning.
@toyowata can you please clarify what 'other targets' you are inferring?

theotherjimmy, 2017-09-20:
@adbridge There might be a bug making all armc6 make exports not supported. **THIS DOES NOT AFFECT ARMC5**

adbridge, 2017-09-20:
@theotherjimmy can we determine if that is actually the case and update this issue accordingly, it can then also be added to the known issues

toyowata, 2017-09-20:
@adbridge Sorry for the confusion.  I have tested with NRF52_DK and K64F only.

cmonr, 2018-07-11:
Closing since exporting using ARMc6 is not supported yet. Once support is added, if this is still an issue, feel free to open a new issue.

closed 2018-07-11 by cmonr