ARMmbed/mbed-os #5145: [OOB 5.6 r2] make_armc5 exporter does not support NUCLEO_F767ZI, NUCLEO_F401RE and DISCO_L476VG

soramame21, 2017-09-20:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F767ZI, NUCLEO_F401RE and DISCO_L476VG|??

**Toolchain:**
ARMC5

**Toolchain version:**
MDK_ARM Professional Ver. : 5.22
C Compiler: v5.06 update 4(build 422)

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed-os-example-fat-filesystem (08607866ff3f09b75e429ac0432587478d5e1fc3)
mbed-os (6e087488e10143222b3b08790eb45600c784a309)

**Expected behavior**
Make can pass to generate a binary executable without error.

**Actual behavior**
Make get failed. please check the attached log.

> "Assemble: irq_cm4f.s"
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 69 (column 8): Error: A1163E: Unknown opcode __FPU_PRESENT , expecting opcode or Macro
>    69 00000030 #ifdef __FPU_PRESENT
>                       ^
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 75: Error: A1167E: Invalid line start 
>    75 0000003c #endif
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 79 (column 8): Error: A1163E: Unknown opcode __FPU_PRESENT , expecting opcode or Macro
>    79 00000040 #ifdef __FPU_PRESENT
>                       ^
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 82: Error: A1167E: Invalid line start 
>    82 0000004a #endif
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 95 (column 8): Error: A1163E: Unknown opcode __FPU_PRESENT , expecting opcode or Macro
>    95 0000005e #ifdef __FPU_PRESENT
>                       ^
> "../mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.s", line 98: Error: A1167E: Invalid line start 
>    98 00000068 #endif
> 6 Errors, 0 Warnings
> make[1]: *** [C:/ioT/testpod/oob5.6/mbed-os-example-fat-filesystem/Makefile:779: mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/irq_cm4f.o] Error 1
> make: *** [makefile:26: all] Error 2

**Steps to reproduce**

> \>mbed import mbed-os-example-fat-filesystem
> \>cd mbed-os-example-fat-filesystem
> \>mbed update mbed-os-5.6-oob2
> \>mbed export -m NUCLEO_F767ZI -i make_armc5 -vv
> \>make
[failed-make-F767ZI.txt](https://github.com/ARMmbed/mbed-os/files/1317102/failed-make-F767ZI.txt)

0xc0170, 2017-09-20:
cc @theotherjimmy

theotherjimmy, 2017-09-20:
BTW, this becomes a problem, because the Arm assebler, `armasm`, only runs the C preprocessor on `.S` files, and not `.s` files. This strikes me as odd behavior, considering that it was originally developed for Windows, and windows does not distinguish....

ashok-rao, 2017-09-20:
@soramame21 .. can you check if https://github.com/ARMmbed/mbed-os/pull/5149 helps ? please confirm..thanks!
cc @MarceloSalazar

theotherjimmy, 2017-09-20:
Hey @ashok-rao. Thanks for mentioning that issue. It **should** have worked, but something is going awry with @soramame21's setup here. We have discussed what that patch brings to the table already, and it did not seem to work.

soramame21, 2017-09-21:
@ashok-rao, @theotherjimmy, cc @MarceloSalazar Thanks for your comments. I had verified that the file is "irq_cm4f.S" although windows treats it as the same as "irq_cm4f.s".
I had applied [the PR #5149](https://github.com/ARMmbed/mbed-os/pull/5149), but doesn't help.
I don't install MinGW or Cygwin in my environment, just using windows command prompt.

> C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem>cd mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/
> 
> C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem\mbed-os\rtos\TARGET_CORTEX\rtx5\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM>dir
>  Volume in drive C is OSDisk
>  Volume Serial Number is 4645-8BCA
> 
>  Directory of C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem\mbed-os\rtos\TARGET_CORTEX\rtx5\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM
> 
> 09/14/2017  10:43    <DIR>          .
> 09/14/2017  10:43    <DIR>          ..
> 09/14/2017  10:43             5,972 irq_cm4f.S
>                1 File(s)          5,972 bytes
>                2 Dir(s)  319,016,480,768 bytes free

I attached the file at here.
[irq_cm4f.zip](https://github.com/ARMmbed/mbed-os/files/1319746/irq_cm4f.zip)

theotherjimmy, 2017-09-21:
PR #5149 will not help with any code issues, as it is purely documentation. I'm not sure why you attached the file. 

The problem is that make is treating the file as if it were named `irq_cm4f.s`. There may be a workaround. Have you tried removing your `mbed-os` directory and deploying again? Windows may be doing something odd when git tells it to rename from `irq_cm4f.s` to `irq_cm4f.S`, and that can be avoided by creating the file with the name `irq_cm4f.S`.

soramame21, 2017-09-21:
@theotherjimmy I'll try to remove mbed-os and deploy it.

theotherjimmy, 2017-09-21:
@soramame21 Turns out Arm Compiler 5's behavior on assembly files is platform dependent! See the `.S` section of [this page](https://developer.arm.com/products/software-development-tools/compilers/arm-compiler-5/docs/dui0472/k/getting-started-with-the-compiler/filename-suffixes-recognized-by-the-compiler)

theotherjimmy, 2017-09-21:
That explains why I can't reproduce this, ever. I'm generally only using UNIX-like systems.

dhwalters423, 2017-09-23:
I am encountering this issue with a KW24D, see log attached
[make_armc5fail.txt](https://github.com/ARMmbed/mbed-os/files/1326859/make_armc5fail.txt)

*Host OS:* Windows 10
*Mbed version:* 5.6 OOB2
*Application:* Blinky
*Export Target:* ARMC5 Make

@theotherjimmy I can confirm the name of the file in my fs is irq_cm4f.S

theotherjimmy, 2017-09-23:
@dhwalters423 yeah, I have a workaround in #5166

PR #5166 (2017-09-25): Force preprocessing of asm for make_armc5
`armasm` seem to preprocess asm with the `.S` extension on linux, but the
windows build of Arm Compiler 5 seems to omit the preprocessing step
unless you have a `.sx` extension. Odd.

Resolves #5145
closed 2017-09-25 by theotherjimmy