ARMmbed/mbed-os #5146: [OOB 5.6 r2] make_gcc_arm exporter does not support NUCLEO_F767ZI and others?

soramame21, 2017-09-20:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F767ZI |??

**Toolchain:**
ARMC5

**Toolchain version:**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed-os-example-fat-filesystem (08607866ff3f09b75e429ac0432587478d5e1fc3)
mbed-os (6e087488e10143222b3b08790eb45600c784a309)

**Expected behavior**
Make can pass to generate a binary executable without error.

**Actual behavior**
Make get failed.

> C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem>make
> C:/ioT/testpod/oob5.6/mbed-os-example-fat-filesystem/Makefile:780: warning: overriding recipe for target '.s.o'
> C:/ioT/testpod/oob5.6/mbed-os-example-fat-filesystem/Makefile:775: warning: ignoring old recipe for target '.s.o'
> "Compile: main.cpp"
> process_begin: CreateProcess(NULL, arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g3 -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=softfp -std=gnu++98 -fno-rtti -Wvla -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DTARGET_STM32F767ZI -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DTARGET_STM32F7 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DARM_MATH_CM7 -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M7 -DDEVICE_RTC=1 -DDEVICE_ANALOGOUT=1 -DTARGET_NUCLEO_F767ZI -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M7 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -D__CORTEX_M7 -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_STM32F767 -DTARGET_STM -DUSBHOST_OTHER -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DMBED_BUILD_TIMESTAMP=1505898471.38 -DDEVICE_ANALOGIN=1 -DTARGET_STM32F767xI -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -include mbed_config.h -I../ -I../. -I.././mbed-os -I.././mbed-os/cmsis -I.././mbed-os/cmsis/TARGET_CORTEX_M -I.././mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_GCC -I.././mbed-os/cmsis/TOOLCHAIN_GCC -I.././mbed-os/drivers -I.././mbed-os/events -I.././mbed-os/events/equeue -I.././mbed-os/features -I.././mbed-os/features/FEATURE_LWIP -I.././mbed-os/features/FEATURE_LWIP/lwip-interface -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F7 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F7/TARGET_NUCLEO_F767ZI -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys/arch -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/api -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv4 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv6 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/priv -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/prot -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp/polarssl -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp/polarssl -I.././mbed-os/features/filesystem -I.././mbed-os/features/filesystem/bd -I.././mbed-os/features/filesystem/fat -I.././mbed-os/features/filesystem/fat/ChaN -I.././mbed-os/features/frameworks -I.././mbed-os/features/frameworks/greentea-client -I.././mbed-os/features/frameworks/greentea-client/greentea-client -I.././mbed-os/features/frameworks/greentea-client/source -I.././mbed-os/features/frameworks/unity -I.././mbed-os/features/frameworks/unity/source -I.././mbed-os/features/frameworks/unity/unity -I.././mbed-os/features/frameworks/utest -I.././mbed-os/features/frameworks/utest/source -I.././mbed-os/features/frameworks/utest/utest -I.././mbed-os/features/mbedtls -I.././mbed-os/features/mbedtls/importer -I.././mbed-os/features/mbedtls/inc -I.././mbed-os/features/mbedtls/inc/mbedtls -I.././mbed-os/features/mbedtls/platform -I.././mbed-os/features/mbedtls/platform/inc -I.././mbed-os/features/mbedtls/platform/src -I.././mbed-os/features/mbedtls/src -I.././mbed-os/features/mbedtls/targets -I.././mbed-os/features/mbedtls/targets/TARGET_STM -I.././mbed-os/features/mbedtls/targets/TARGET_STM/TARGET_STM32F7 -I.././mbed-os/features/nanostack -I.././mbed-os/features/netsocket -I.././mbed-os/features/netsocket/cellular -I.././mbed-os/features/netsocket/cellular/generic_modem_driver -I.././mbed-os/features/netsocket/cellular/utils -I.././mbed-os/features/storage -I.././mbed-os/hal -I.././mbed-os/hal/storage_abstraction -I.././mbed-os/platform -I.././mbed-os/rtos -I.././mbed-os/rtos/TARGET_CORTEX -I.././mbed-os/rtos/TARGET_CORTEX/rtx4 -I.././mbed-os/rtos/TARGET_CORTEX/rtx5 -I.././mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7 -I.././mbed-os/rtos/TARGET_CORTEX/rtx5/TARGET_RTOS_M4_M7/TOOLCHAIN_GCC -I.././mbed-os/targets -I.././mbed-os/targets/TARGET_STM -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7 -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/TARGET_NUCLEO_F767ZI -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/device -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/device/TOOLCHAIN_GCC_ARM -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F7/device -o main.o ../main.cpp, ...) failed.
> make (e=2): The system cannot find the file specified.
> make[1]: *** [C:/ioT/testpod/oob5.6/mbed-os-example-fat-filesystem/Makefile:792: main.o] Error 2
> make: *** [makefile:26: all] Error 2
> 

**Steps to reproduce**

> \>mbed import mbed-os-example-fat-filesystem
> \>cd mbed-os-example-fat-filesystem
> \>mbed update mbed-os-5.6-oob2
> \>mbed export -m NUCLEO_F767ZI -i make_gcc_arm -vv
> \>make

0xc0170, 2017-09-20:
cc @bcostm @LMESTM @adustm @jeromecoutant 

@theotherjimmy

theotherjimmy, 2017-09-20:
@soramame21 Please add `arm-none-eabi-gcc` and friends to your path.

soramame21, 2017-09-20:
@theotherjimmy what do you mean?

theotherjimmy, 2017-09-20:
You need to be able to run the executable `arm-none-eabi-gcc` and other executables named `arm-none-eabi-*` without specifying the full path to that executable. To do this add the location of these executables to your PATH environment variable.

soramame21, 2017-09-20:
@theotherjimmy yes, I added my PATH setting in the description.

theotherjimmy, 2017-09-20:
Did you make sure to restart the terminal you were using after setting up your path?

theotherjimmy, 2017-09-20:
Odd, If that works, then the makefile should work from the same terminal

soramame21, 2017-09-20:
I didn't change PATH setting for gcc. because it works well so far. now I verified and found 2 versions of arm-none-eabi-gcc are installed. my environment variable are as following.

> MBED_GCC_ARM_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\5.4 2016q3\bin
> PATH=;;; C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin;...

The 2nd version (6-2017-q2-update) is valid on command line.

theotherjimmy, 2017-09-20:
@soramame21 I really don't know what's going wrong with your setup. I can't reproduce the issue on my machine. Though my machine does use Linux...

theotherjimmy, 2017-09-20:
@soramame21 Maybe the space in front of the `;` in the path? On the other hand, If you can run the compiler from the command line, I don't know why make can't run the compiler.

soramame21, 2017-09-20:
@theotherjimmy the following path included in the generated Makefile is correct??
"../" must be removed.
> INCLUDE_PATHS += -I../
> INCLUDE_PATHS += -I../.
> INCLUDE_PATHS += -I.././DebugConfig
> INCLUDE_PATHS += -I.././mbed-os
> INCLUDE_PATHS += -I.././mbed-os/cmsis
> INCLUDE_PATHS += -I.././mbed-os/cmsis/TARGET_CORTEX_M
> INCLUDE_PATHS += -I.././mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_GCC
> INCLUDE_PATHS += -I.././mbed-os/cmsis/TOOLCHAIN_GCC
> INCLUDE_PATHS += -I.././mbed-os/drivers
> INCLUDE_PATHS += -I.././mbed-os/events
> INCLUDE_PATHS += -I.././mbed-os/events/equeue
> INCLUDE_PATHS += -I.././mbed-os/features

theotherjimmy, 2017-09-20:
> the following path included in the generated Makefile is correct??

Those paths are correct

> "../" must be removed.

`../` must **not** be removed.

The garbage looking stuff at the top of the makefile re-runs make in the build directory, required the `../` prefix on include paths.

theotherjimmy, 2017-09-20:
> `INCLUDE_PATHS += -I.././DebugConfig`

This looks like you did one export after another without cleaning between. That is **not** supported, and you should make sure to clean up your directories after doing an export.

theotherjimmy, 2017-09-20:
@soramame21 Further, those issues that you mention have nothing to do with the error you are seeing. 
```
make (e=2): The system cannot find the file specified.
```
Is the error you get when you can't run the compiler. If it was complaining about not finding an include file, you would get a gcc-style error as well.

soramame21, 2017-09-20:
I modified tools/export/GettingStarted.html following the pull reuqest below, but still get the same error.
https://github.com/ARMmbed/mbed-os/pull/5149/files

theotherjimmy, 2017-09-20:
@soramame21 Yeah, that's just a documentation thing. It's not going to affect the Makefile at all.

soramame21, 2017-09-20:
@theotherjimmy I uploaded my generated Makefile, Could you verify it please?

theotherjimmy, 2017-09-20:
@soramame21 I don't know what you want me to verify about it. It looks like an exported makefile, and nothing catches my eye as wrong.

soramame21, 2017-09-20:
@theotherjimmy I ran following command and got the same error.

> C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem\BUILD>arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -o main.o ../main.cpp
> arm-none-eabi-g++: error: CreateProcess: No such file or directory

soramame21, 2017-09-20:
once I removed ../main.cpp, then I got the expected error.

> C:\ioT\testpod\oob5.6\mbed-os-example-fat-filesystem\BUILD>arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -o main.o
> arm-none-eabi-g++: fatal error: no input files
> compilation terminated.

theotherjimmy, 2017-09-20:
@soramame21 Thanks for the debug info. That's very strange.

theotherjimmy, 2017-09-20:
Oh, That might be related to the fact that GCC will execute the preprocessor/assembler in a process...

You might want to check that the arm-none-eabi-* binutils are in your path.

I think that you need:
 - `arm-none-eabi-as` (asseblmer)
 - `arm-none-eabi-cpp` (preprocessor)
 - `arm-none-eabi-g++` (c++ compiler)
 - `arm-none-eabi-c++` (also the c++ compiler)
 - `arm-none-eabi-gcc` (the frontend to all of the others)
 - `arm-none-eabi-ld` (the linker)

There might be a few more, let me llook and get back to you.

theotherjimmy, 2017-09-20:
Looking about the internet, I have seen reports that you may need to run `make` from within the appropriate Mingw or Cygwin environment if that's where you installed the arm-none-eabi-gcc toolchain.

theotherjimmy, 2017-09-20:
Also according to various reports from all over the internet, a reboot may solve it.

JojoS62, 2017-09-20:
it looks like a gcc installation problem, here are a lot of reports: https://stackoverflow.com/questions/3848357/createprocess-no-such-file-or-directory
Do you have some other MinGW or Cygwin stuff installed and in the environment? That may interfere.

soramame21, 2017-09-21:
@theotherjimmy, @JojoS62 Thanks for your comments. I don't install MinGW or Crgwin on my environment so far. just working on windows command prompt.

soramame21, 2017-09-26:
@theotherjimmy, @JojoS62 I can build the example after reinstalling Gnu toolchain, and passed the test. thank you for your support, I close this issue.
cc @MarceloSalazar

closed 2017-09-26 by soramame21