ARMmbed/mbed-os #5153: Memap stats depth of 0 does unexpected behavior

theotherjimmy, 2017-09-20:
# Description
- Type: Bug and/or Enhancement  -- I'm honestly not sure
- Related issue: `#abc`
- Priority: Minor

I don't think that `mbed compile` or `mbed test` for the argument `--stats-depth 0` does what I expect. With `--stats-depth 2` you get the following output:
```
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |    90 |     3 |   41 |
| [lib]/libc.a     | 24469 |  2472 |   56 |
| [lib]/libgcc.a   |  7134 |     0 |    0 |
| [lib]/libnosys.a |    32 |     0 |    0 |
| [lib]/misc       |   208 |     8 |   25 |
| blinky/BUILD     | 35933 |   273 | 6010 |
| fat-fs/BUILD     |  2811 |     0 | 1960 |
| mbed-os/BUILD    |  2036 |     4 |  180 |
| Subtotals        | 72713 |  2760 | 8272 |
+------------------+-------+-------+------+
```
Disregarding the bugs mentioned in #5124 #5127 #4976, This output makes sense: it truncates the hierarchy from the file system at a depth of 2. The simple `du` (disk usage) comand does the same thing with it's `--max-depth` argument.
For reference,
```
# du --max-depth 2 -h
0	./.git/branches
44K	./.git/hooks
4.0K	./.git/info
2.4M	./.git/refs
223M	./.git/objects
2.5M	./.git/logs
229M	./.git
8.0K	./.github
8.0K	./TESTS/host_tests
4.0K	./TESTS/integration
84K	./TESTS/mbed_drivers
72K	./TESTS/mbed_functional
116K	./TESTS/mbed_hal
36K	./TESTS/mbedmicro-mbed
100K	./TESTS/mbedmicro-rtos-mbed
44K	./TESTS/storage_abstraction
12K	./TESTS/events
12K	./TESTS/mbedtls
4.0K	./TESTS/netsocket
12K	./TESTS/network
504K	./TESTS
4.1M	./features/FEATURE_BLE
1.4M	./features/FEATURE_COMMON_PAL
8.1M	./features/FEATURE_UVISOR
488K	./features/frameworks
3.5M	./features/mbedtls
2.6M	./features/storage
5.2M	./features/FEATURE_LWIP
40K	./features/TESTS
364K	./features/filesystem
79M	./features/nanostack
284K	./features/netsocket
14M	./features/unsupported
118M	./features
780K	./rtos/TARGET_CORTEX
876K	./rtos
3.0M	./tools/arm_pack_manager
156K	./tools/bootloaders
56K	./tools/compliance
84K	./tools/config
24K	./tools/data
24K	./tools/dev
2.1M	./tools/export
364K	./tools/host_tests
16K	./tools/misc
[... 202 lines cut ...]
| mbed-os/BUILD/tests/nrf51_dk/gcc_arm/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/device/TOOLCHAIN_GCC_ARM/startup_NRF51822.o |   260 |     0 |    0 |
| mbed-os/BUILD/tests/nrf51_dk/gcc_arm/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_SDK11/softdevice/common/softdevice_handler/softdevice_handler.o  |    12 |     0 |    9 |
| Subtotals                                                                                                                                      | 72713 |  2760 | 8272 |
+------------------------------------------------------------------------------------------------------------------------------------------------+-------+-------+------+
```

Again, please disregard the aforementioned bugs.

and `du`:
```
# du --max-depth 0 -h
692M	.
```

For reference, I think that `--stats-depth 0` should print:
```
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Subtotals | 72713 |  2760 | 8272 |
+-----------+-------+-------+------+
Total Static RAM memory (data + bss): 11032 bytes
Total Flash memory (text + data): 75473 bytes
Image: BUILD/tests/nrf51_dk/gcc_arm/TESTS/events/queue/queue.hex
```

0xc0170, 2017-09-21:
cc @MarceloSalazar

MarceloSalazar, 2017-09-21:
> Memap and du diverge on what to do when --stats-depth/--map-depth are set to 0.

Really good catch :)
The `--stats-depth` feature in memap wasn't introduced with the intention to have similarities with `du`, but I don't see a problem with moving towards this.

Question: how do you suggest we display a full report without using `0`?
IMO having a full report after each compile seems too much information for the user.

Nodraak, 2017-10-02:
I believe `0` should give a result similar to `du`. If a user wants the full report they can use `99` or whatever bug number they can think of. My 2 cents :)

PR #7377 (2018-07-05): Tools: Summarize stats when depth is 0
### Description

#5153 describes an odd inconsistency in memap behavior, or at least where
my expectations were let down. It would print the entire memory map at full
depth when the depth was specified as 0. This PR changes that to print 
something like:

```
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Subtotals | 38450 |  2564 | 7052 |
+-----------+-------+-------+------+
```

Resolves #5153 

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] [... cut]
closed 2018-07-05 by cmonr