ARMmbed/mbed-os #5156: How to store the LWM2M security object from bootstrap in flash?

amq, 2017-09-21:
## Question

**How to?**

1. Device connects to a bootstrap server without using authentication
2. Receives a security object
3. Saves it permanently
4. Subsequent connections to the LWM2M server are using authentication, even after a reboot

It can be done in a vendor-specific way (EEPROM / flash), but I'm looking for a universal solution, also regarding the memory mapping. I imagine [Flash IAP](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/flash/) can be used for this, but I'm not sure how to store the security object properly.

0xc0170, 2017-09-21:
cc @JanneKiiskila @yogpan01 @geky

geky, 2017-09-21:
From the storage side, your may be able to use this guy to wrap FlashIAP objects in a block device. Then you can use it just like external storage: https://github.com/ARMmbed/flashiap-driver

However, the flashiap-driver is experimental right now. A big concern is that flashiap may freeze the processor for long periods of time as it writes to flash. I'm not off this will interfere with network connections and the rest of the stack.

amq, 2017-09-21:
@geky, seems like it's private.

geky, 2017-09-21:
Ah sorry about that, it should be public now.

Also as @marcuschangarm pointed out, you can end up wearing out your flash if you're not careful. The existing FAT filesystem in mbed OS is poorly suited for internal flash. There is https://github.com/armmbed/mbed-spiffs and https://github.com/armmbed/mbed-littlefs that are filesystems with wear-leveling, but I'm not sure how difficult it will be to adopt with the higher layers. We're currently working on this.

0xc0170, 2017-10-13:
@amq Is this question still unanswered or resolved?

amq, 2017-10-13:
@0xc0170 answered, but looks like the solution is not fully there yet

closed 2017-10-13 by amq