ARMmbed/mbed-os #5167: mbed-os-blinky-example export to gnuarmeclipse does not program

vergil-SI, 2017-09-22:
### Description
- Type: Bug/Question
- Related issue: #5134 ?

---------------------------------------------------------------
## Bug

**Target**
NRF52840-Preview-DK (NRF52840_DK)

**Toolchain:**
GCC_ARM

**Toolchain version:**
Have tested both:
gcc-arm-none-eabi-6-2017-q1-update
gcc-arm-none-eabi-6-2017-q2-update

**mbed-cli version:** 
1.2.0

**DAPLink version:**
0243
**Expected behavior**
Board Programs with blinking LED1
**Actual behavior**
Board does not program, dropped binary/hex/elf file is not recognized, and DAPLINK drive returns FAIL.TXT
"The transfer timed out."

**Steps to reproduce**

mbed import mbed-os-blinky-example
mbed target NRF52840_DK
mbed export gnuarmeclipse
Build the project in eclipse.
Project builds with no errors.
binary and elf file are created (also attempted changing build artifact to .hex)
DAPLINK times out when any of these are drag and dropped into the drive.

Other notes:
mbed-cli compiles and programs perfectly before attempting to export the project.

Any assistance getting this thing to program would be appreciated.

0xc0170, 2017-09-22:
cc @nvlsianpu @anangl @pan-

pan-, 2017-09-22:
@vergil-SI Could you indicate what is your operating system ? 
Might be related to https://github.com/mbedmicro/DAPLink/issues/316 .

vergil-SI, 2017-09-22:
I'm almost positive I'm running Ubuntu 16.04.3 LTS. If the version is of high significance, I'd need to verify that in a few hours when I have access to the machine again just to be sure..

theotherjimmy, 2017-09-22:
@vergil-SI Do not flash through copy to mass storage. Instead please use the PyOCD plugin. 

Even if your DAPLink had not refused, you would not have a soft device on the board, and it would not boot.

dhwalters423, 2017-09-22:
@vergil-SI As @pan- said I believe you may be experiencing the exact symptoms reported in mbedmicro/DAPLink#316. I can reproduce on my machine with the same set up that you have.

vergil-SI, 2017-09-23:
Thanks for the PyOCD soft device advice. That actually probably explains a separate issue I had previously. Glad to know I'm not the only one with this issue.

0xc0170, 2017-10-13:
@vergil-SI I  believe this is related to DAPLink, the latest answer can be found - https://github.com/mbedmicro/DAPLink/issues/323, the target is not supported

cc @nvlsianpu @anangl

I would close this issue, and follow up in the 2 referenced issues here

JojoS62, 2017-10-13:
There is also another problem with the nrf Targets and the exporters: the nrf targets need a precompiled runtime (closed source part) that is delivered as a .hex file and cannot be included in the link process. So the compiled project and the Nordix hexfile must be concatenated with a tool as postbuild step. A workaround is to flash the S130.hex manually before the project.hex, that worked for me.
Even if the BLE functions are not used, the app is linked to start at a higher address and the S130 must be present to jump to the mbed start address.

vergil-SI, 2017-10-13:
Good info, thanks. I'll close this and follow the referenced issues.

closed 2017-10-13 by vergil-SI

edablan, 2018-07-09:
I think this issue is related to what has been talked about here. 

I have a UBLOX NINA-B1 (it uses the nRF52832 chipset). This micro is especially designed for people  needing a supported BLE stack (nordic SDK). I have cloned the mbed-os-example-ble repo, and then chosen the BLE_HearRate. I clone the latest mbed-os repo into this direcotry example,  export to IAR, and MTB_UBLOX_NINA_B1 target name. I can open the created workspace, build, and even go into debug mode. While I step through in the main function everything works, except when it is time to initialize the ble peripheral and thus enable and start the SoftDevice I end up getting a hardfault. On futher inspection of the linker configuration file (.icf) it seems it is setup by default for no softdevice:

`/* Default to no softdevice */
if (!isdefinedsymbol(MBED_APP_START)) {
    define symbol MBED_APP_START = 0x0;
}`

At the bottom of the .icf file this is written:

`/*This is used for mbed applications build inside the Embedded workbench
Applications build with the python scritps use a hex merge so need to merge it
inside the linker. The linker can only use binary files so the hex merge is not possible
through the linker. That is why a binary is used instead of a hex image for the embedded project.
*/
if(isdefinedsymbol(SOFT_DEVICE_BIN))
{
  place at address mem:__code_start_soft_device__ { section .noinit_softdevice };
}`

However, the SOFT_DEVICE_BIN is not defined anywhere in the .icf file. I also can't find the softdevice binary file in any of the mbed-os directories specifically the S132. Obviously this can be downloaded, but not sure where to put the .bin file for it to do the merge.

Though no workarounds are mentioned, it seems anyone using the nordic nRF52x chips won't be able to actually use them with nordic SoftDevice running mbed and building in IAR currently? Any suggestions or workarounds would be great. At this point I'm dead in the water.