ARMmbed/mbed-os #5168: For target target=ARM_BEETLE_SOC and toolchain=ARM/console build time is high

studavekar, 2017-09-22:
### Description

For target target=ARM_BEETLE_SOC and toolchain=ARM/console build time is high about 28 mins, This is causing a slow down in CI pipeline.

- Type: Bug 
- Priority:  Major 

**Target**
ARM_BEETLE_SOC 

**Toolchain:**
IAR

0xc0170, 2017-09-22:
cc @abhishek-pandit @mmorenobarm

pan-, 2017-09-22:
Is it related to flashing time ? IIRC it was taking a lot of time to flash this target. 
@c1728p9 Is there any constraints on flashing throughput in the DAP link specification ?

c1728p9, 2017-09-22:
@pan- nope, flashing is limited by the flash erase and write times of the device.

0xc0170, 2017-09-28:
> @abhishek-pandit @mmorenobarm

bump

abhishek-pandit, 2017-09-28:
Adding @tkaman to the issue

abhishek-pandit, 2017-09-28:
@studavekar @0xc0170 

We will check it out.
Beetle code hasn't changed in a while and the platform is the same, so I am wondering what's changed recently that's caused this problem? Any idea?
Thanks

tkaman, 2017-09-29:
I will look into this and hope to have an update soon

tkaman, 2017-10-04:
I cannot reproduce the issue on my machine, for me the given mbed test command with ARM_BEETLE_SOC and IAR goes through in about ~2 minutes (this is even a minute faster then with ARMCC).
On the other hand in the description the reproduce command does not contain ARM_BEETLE_SOC as target, a Realtek chip is the target there.
I tried this target and this is really slow, this takes about 45-50 minutes with IAR, while with ARMCC the build is only 3-4 minutes.
I've built both targets with both compilers many times, so this is not based on one sample.

Could you please check again whether you really see the issue with Beetle, and if yes, could you give some more details so that I can reproduce?

Thanks

studavekar, 2017-10-05:
@tkaman you're right in above comment, i am closing this issue.

closed 2017-10-05 by studavekar