ARMmbed/mbed-os #5170: Malformed input when parsing ARMCC map

teetak01, 2017-09-22:
### Description
- Type: Bug
- priority: minor

**Target**
K64F + 6LOWPAN
UBLOX+WIFI

**Toolchain:**
ARMCC 5.06 (update 3)
**mbed-os**
5.5.7

```
00:43:30.349 [K64F_6LOWPAN ARM] Malformed input found when parsing ARMCC map: libnanostack.ar(rpl_mrhof.o)
00:43:30.349 [K64F_6LOWPAN ARM] Malformed input found when parsing ARMCC map: libnanostack.ar(adaptation_interface.o)
00:43:30.349 [K64F_6LOWPAN ARM] Malformed input found when parsing ARMCC map: libnanostack.ar(cipv6_fragmenter.o)
```

Same for Ublox_EVK ARMCC

```
00:24:14.878 [UBLOX_EVK_ODIN_W2_WIFI4 ARM] Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_aes.o)
00:24:14.878 [UBLOX_EVK_ODIN_W2_WIFI4 ARM] Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(controller_startup_WL18.o)
00:24:14.878 [UBLOX_EVK_ODIN_W2_WIFI4 ARM] Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_uart.o)
```

Looks ugly, but does not seem to affect code.

teetak01, 2017-09-22:
@SeppoTakalo @andreaslarssonublox

0xc0170, 2017-09-22:
cc @theotherjimmy

0xc0170, 2017-09-22:
This is duplicate, please retest and close or we reopen the old one I referenced above

teetak01, 2017-09-22:
Nice, seems to be fixed in Mbed OS 5.6-RC3.

closed 2017-09-22 by teetak01