ARMmbed/mbed-os #5185: LowPowerTicker: Inconsistent callback calls when using multiple ticker objects

maciejbocianski, 2017-09-25:
### Description
- Type: Bug
- Related issue: `#5150`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
GCC_ARM -- 6.3.1,
ARM -- 5.24 (Flex) ARM Compiler 5.06 update 5 (build 528),
IAR -- ANSI C/C++ Compiler V7.80.4.12462/W32 for ARM

**mbed-os sha:**
c6f655c02

**Expected behavior**
All callbacks fired 23 times in 100 ms intervals

**Actual behavior**
Not all callbacks fired 23 times
Some callbacks fired in time `start + 2s` 

**Code**
```

#define TICKER_COUNT 16

volatile bool ticker_callback_flag;
volatile uint32_t multi_counter;

struct CalbackStruct
{
    uint32_t time;
    uint32_t lp_time;
    uint32_t fired;
    void (*fun)(void);
    LowPowerTicker *ticker;
};

template <uint32_t id>
void wait_callback_multi(void);

volatile CalbackStruct callbacks[TICKER_COUNT] =
{
    { 0, 0, 0, wait_callback_multi<0>, NULL },
    { 0, 0, 0, wait_callback_multi<1>, NULL },
    { 0, 0, 0, wait_callback_multi<2>, NULL },
    { 0, 0, 0, wait_callback_multi<3>, NULL },
    { 0, 0, 0, wait_callback_multi<4>, NULL },
    { 0, 0, 0, wait_callback_multi<5>, NULL },
    { 0, 0, 0, wait_callback_multi<6>, NULL },
    { 0, 0, 0, wait_callback_multi<7>, NULL },
    { 0, 0, 0, wait_callback_multi<8>, NULL },
    { 0, 0, 0, wait_callback_multi<9>, NULL },
    { 0, 0, 0, wait_callback_multi<10>, NULL },
    { 0, 0, 0, wait_callback_multi<11>, NULL },
    { 0, 0, 0, wait_callback_multi<12>, NULL },
    { 0, 0, 0, wait_callback_multi<13>, NULL },
    { 0, 0, 0, wait_callback_multi<14>, NULL },
    { 0, 0, 0, wait_callback_multi<15>, NULL }
};

template <uint32_t id>
void wait_callback_multi(void)
{
    if(!callbacks[id].fired) {
        callbacks[id].time = us_ticker_read();
        callbacks[id].lp_time = lp_ticker_read();
        //callbacks[id].ticker->detach();
    }
    callbacks[id].fired++;
    multi_counter++;
}

int main()
{
	LowPowerTicker ticker[TICKER_COUNT];

	uint32_t start = us_ticker_read();
	uint32_t lp_start = lp_ticker_read();
	multi_counter = 0;
	for (int i = 0; i < TICKER_COUNT; i++) {
		callbacks[i].ticker = &ticker[i];
		ticker[i].attach_us(callback(callbacks[i].fun), 100 * 1000);
	}

	wait_us(2350*1000);
	for (int i = 0; i < TICKER_COUNT; i++) {
		ticker[i].detach();
	}
	printf("start time: %u lp_time: %u\r\n", start, lp_start);
	printf("ID|FIRED|TIME|LPTIME\r\n");
	for (int i = 0; i < TICKER_COUNT; i++) {
		printf("%2i %5u %10u %10u\r\n", i, callbacks[i].fired, callbacks[i].time, callbacks[i].lp_time);
	}
    printf("multi_counter: %d\r\n", multi_counter);
    printf("\r\n");
    printf("*******************************************************************************\r\n");
    printf("\r\n");
}
```

```
start time: 0 lp_time: 2242955688
ID|FIRED|TIME|LPTIME
 0    22      99996 2243055694    // OK
 1    22     100010 2243055694    // OK
 2    22     100030 2243055725    // OK
 3    21    2099732 2245055755    // ????
 4    23    2099742 2245055755
 5    23    2099751 2245055786
 6    23    2099761 2245055786
 7    23    2099770 2245055786
 8    23    2099780 2245055816
 9    23    2099789 2245055816
10    23    2099799 2245055816
11    23    2099808 2245055816
12    23    2099818 2245055847
13    23    2099827 2245055847
14    23    2099837 2245055847
15    23    2099846 2245055877
multi_counter: 363

```

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-09-26:
cc @mmahadevan108

0xc0170, 2017-10-02:
@mmahadevan108 are you able to reproduce this ?

mmahadevan108, 2017-10-02:
Not yet, I will take a look at this shortly.

0xc0170, 2017-10-06:
> Not yet, I will take a look at this shortly.
 
Any luck?

mmahadevan108, 2017-10-09:
I am able to reproduce the issue and will look at fix.

mmahadevan108, 2017-10-13:
PR https://github.com/ARMmbed/mbed-os/pull/5314 should fix this issue.

maciejbocianski, 2017-10-16:
Fixed works properly
```
start time: 0 lp_time: 560159393
ID|FIRED|TIME|LPTIME
 0    23     100003  560259399
 1    23     100059  560259460
 2    23     100068  560259460
 3    23     100078  560259490
 4    23     100088  560259490
 5    23     100098  560259490
 6    23     100118  560259521
 7    23     100133  560259521
 8    23     100148  560259552
 9    23     100163  560259552
10    23     100193  560259582
11    23     100218  560259613
12    23     100234  560259643
13    23     100249  560259643
14    23     100270  560259674
15    23     100285  560259674

```

maciejbocianski, 2017-10-23:
fix moved to https://github.com/ARMmbed/mbed-os/pull/5347

maciejbocianski, 2017-10-23:
fixed since release-version: 5.6.4

closed 2017-10-23 by maciejbocianski