ARMmbed/mbed-os #5190: NUCLEO-F429ZI flash program bug

Alonof, 2017-09-25:
### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-F429ZI

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
db4be9469 (HEAD, tag: mbed_lib_rev149, tag: mbed-os-5.5.5, tag: latest) Merge pull request #4889 from ARMmbed/release-candidate

**Expected behavior**
when trying to program to in intranel flash using the flashIAP APIs ` FlashIAP` ,
The program function verifies that the size is aligned to page_size.
but the page size given from the API `get_page_size()` is hardcoded to 0x4000.
This can be found in `mbed-os\targets\TARGET_STM\TARGET_STM32F4\flash_api.c`
according to the STM SDK and HAL function, the flash can be program to 8,16,32 & 64 bits

**Actual behavior**
Can write to flash small buffer under 0x4000 length which is the wrong behavior 

**Steps to reproduce**
int flash
program to flash buffer with size less than 0x4000

bcostm, 2017-09-26:
Please use mbed-os latest version. The flash get size has been updated in PR #4980
Thanks

0xc0170, 2017-09-26:
@Alonof PLease test, and close if resolved

Alonof, 2017-09-28:
Hi
Great work (and very fast) thanks
Alon

closed 2017-09-28 by Alonof