ARMmbed/mbed-os #5195: [OOB 5.6 r3]  LPC54114 fails to boot up with GCC compiler

maclobdell, 2017-09-25:
### Description
- Type: Bug 
- Related issue: n/a
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
LPC54114

**Toolchain:**
GCC_ARM - Fails
ARM - Pass
IAR - Pass

**Toolchain version:**
GCC 6-2017-q1-update

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed-os-5.6  

**DAPLink version:**
n/a

**Expected behavior**
  mbed-os-example-blinky main.cpp results in blinking led.

**Actual behavior**
led does not blink when compiling, downloading, reset LPC54114 with GCC compiler.  Other compilers work fine.  

Suspect Kernel error when RTOS is initialized before main. 

**Steps to reproduce**
import mbed-os
change branch to mbed-os-5.6  
add main.cpp from blinky program.
compile and test.

c1728p9, 2017-09-25:
I created #5197 to fix this

0xc0170, 2017-10-13:
@maclobdell The fix was integrated. Please reopen if the issue persist

closed 2017-10-13 by 0xc0170