ARMmbed/mbed-os #5206: Efficiently wait for accelerator operations in HAL

andresag01, 2017-09-27:
### Description
- Type: Enhancement

----------------------------------------------------------------
## Enhancement
When looking at the code for hardware acceleration, we noticed that there are many locations where a command is issued to the accelerator, then the thread must must wait for the operation to be completed. The mbed TLS API is mostly synchronous, so for lack of an alternative mechanism, the thread simply is set to busy wait until the accelerator has finished. This behaviour is inefficient as those CPU cycles could be used to execute another thread instead of manually checking on a condition. 

**Suggested enhancement**
An idea to avoid the busy waits mentioned above is to have some mechanism that enabled HAL code to issue a command for the accelerator or a peripheral and then wait on a condition that would be satisfied (e.g. by means of a semaphore) when the operation has completed.

**Pros**
Threads can more efficiently be scheduled and blocking in the form of busy waiting for events to occur can be avoided in a much easier way.

**Cons**
Probably there might be issues with backwards compatibility when adding this new feature.

bulislaw, 2017-10-02:
That's a very good point, I would say we should think about using RTOS in HAL, but that cannot happen as long as we support mbed 2.
Maybe we can temporarily introduce a wrapper for semaphore that would expand to RTOS sempahore if present and to busy loop for mbed 2?

@sg- @c1728p9 @0xc0170

0xc0170, 2017-10-06:
I would rather keep HAL API simple and build on top of it  - provide functionality to allow to be running in background.

@andresag01 Where in HAL is this? Can you pinpoint to our code base?

andresag01, 2017-10-06:
@0xc0170: The crypto hardware acceleration code is full of busy wait loops while we wait for the accelerator to complete commands. For example: [here](https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/targets/TARGET_STM/aes_alt.c#L156). Also consider looking at the open PRs for crypto hardware accelerator.

pan-, 2017-10-06:
> I would rather keep HAL API simple and build on top of it - provide functionality to allow to be running in background.

That wouldn't solve the issue with busy waiting i.e wasted CPU cycles which can be use for other processing or save power.

@bulislaw The event middleware offers an interesting solution which avoid wasting CPU cycles with or without the RTOS present (see [here](https://github.com/ARMmbed/mbed-os/blob/master/events/equeue/equeue_mbed.cpp))

loverdeg-ep, 2017-10-06:
> The event middleware offers an interesting solution which avoid wasting CPU cycles with or without the RTOS present (see here)

+1 for anything that brings us back closer to the event driven dream

pan-, 2017-10-06:
@loverdeg-ep The link I've posted is an implementation of a *semaphore* with or without the RTOS. I wasn't advocating for the direct use of the event middleware to solve that issue. Apologies for the unclear message.

I don't think the use of the event middleware would be a sensible decision at the HAL level. Of course it would be feasible to have asynchronous API at that level however, due to the synchronous nature of mbed tls functions, the glue sitting between mbed tls and the HAL would have to act like a semaphore/condition variable.

loverdeg-ep, 2017-10-06:
@pan- Thanks for clarification; I wasn't necessarily implying that it be used directly either.
More so implying that preference is to keep things asynchronous and RTOS as optional as possible.

0xc0170, 2019-06-03:
Referencing here latest work that my help to fix this issue: https://github.com/ARMmbed/mbed-os/pull/10104

40Grit, 2019-06-03:
Glad to see this bumped

@trowbridgec

RonEld, 2019-06-11:
Also, the already existing  `PlatformMutex` feature can be used to synchronise as well

40Grit, 2019-06-11:
@geky any application for your asynchronizer thingy here?

https://github.com/geky/coru

40Grit, 2019-06-16:
Reason for closure

@linlingao

linlingao, 2019-06-17:
Looks like @oliverjharper closed it. Oliver?

oliverjharper, 2019-06-17:
@linlingao I closed the IOTCRYPT ticket, Jira has mirrored the closure here as well. @Patater and the crypto team are not currently working on this. Perhaps the Jira ticket should be re-opened and moved to the most appropriate Jira project?

linlingao, 2019-06-17:
I'm reopening this issue here. @oliverjharper Let's work out Jira handling offline.

closed 2019-10-31 by ciarmcom