ARMmbed/mbed-os #5209: LPC1768 mbed_interface_connected not functioning correct on Mbed OS 5

janjongboom, 2017-09-27:
### Description

- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
LPC1768

According to [this report](https://os.mbed.com/questions/79090/mbed_interface_reset-with-OS5-not-resett/) `mbed_interface_connected()` no longer reports the right value, so detecting if the interface is connected does not work...

0xc0170, 2017-09-27:
cc @c1728p9 @toyowata

toyowata, 2017-09-29:
@janjongboom @c1728p9 

I quick tested this with following test code.
```cpp
#include "mbed.h"

DigitalOut led(LED1, 0);

int main() {
    printf("mbed_interface_connected : %d\n", mbed_interface_connected()); 
    while(1) {
        led = !led;
        wait(0.2);
    }
}
```

And I run the test code on ```mbed-os-5.5.6``` and ```mbed-os-5.6.0``` release tag tree with ARM and GCC_ARM build.   I got following result for all combination above.

```
mbed_interface_connected : 1
```

So, the ```mbed_interface_connected``` function itself works as expected.

janjongboom, 2017-09-29:
Interesting. I've reported it back to the user.

closed 2017-09-29 by janjongboom

lucasec, 2018-07-27:
While this may be an unrelated issue, I have found that mbed_interface_connected returns 0 when called shortly after bootup on the LPC1768 platform, firmware 141212.

The first call if made early in main() returns 1, but subsequent calls return 0. For example:

```
int main() {
    while (1) {
       printf("Connected: %d\r\n", mbed_interface_connected());
       wait(1);
    }
}
```

Prints:
Connected: 1
Connected: 0
Connected: 0
Connected: 0
...