ARMmbed/mbed-os #5211: OdinWiFiInterface constructor has side-effects

kylestein-arm, 2017-09-27:
### Description
The construction of an instance of OdinWiFiInterface manipulates the system event queue in such a way that previous configuration is lost.  In particular, when used with MbedCloudClient, the OdinWiFiInterface can cancel timers that were set up by the cloud client.

- Type: Bug
- Related issue: none known
- Priority: Major

---------------------------------------------------------------
## Bug
Code behaves differently depending on when an OdinWiFiInterface is instantiated.  
For instance
```
{
    OdinWiFiInterface wifi;
   
    ...

    MbedCloudClient _cloud_client;

    _cloud_client.setup(&wifi);
    ...
}
```
runs as expected,
But
```
{
    OdinWiFiInterface *wifi;
    MbedCloudClient _cloud_client;

    ...
    wifi = new OdinWiFiInterface();
    ...
     _cloud_client.setup(wifi);
    ...
}
```
results in a cloud client whose timers do not fire (for example it won't send keep-alive messages to the cloud)

Files using mbed-cloud-client-example-restricted to demonstrate this behavior are attached.

[original.log](https://github.com/ARMmbed/mbed-os/files/1337375/original.log)
original.log - the trace debug log of mbed-cloud-client-example-restricted (unmodified). 
[easy-connect-patch.txt](https://github.com/ARMmbed/mbed-os/files/1337383/easy-connect-patch.txt)
easy-connect-patch.txt - patch for easy-connect.h
[modified.log](https://github.com/ARMmbed/mbed-os/files/1337384/modified.log)
modified.log - the trace debug log of mbed-cloud-client-example-restricted (with easy-connect.patch applied)

mbed-cloud-client-example-restricted uses the easy-connect library which instantiates a global OdinWiFiInterface at initialization.  The patch modifies easy-connect, so that the OdinWiFiInterface is instantiated dynamically just before it is needed.

original.log shows the mbed cloud-client sending a keep-alive message as the result of a timer expiration
...
[DBG ][mClt]: M2MNsdlInterface::timer_expired - M2MTimerObserver::Registration - Send update registration
[DBG ][mClt]: M2MNsdlInterface::send_update_registration( lifetime 0)
...

These log messages are missing from modified.log.

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.1.0

**mbed-os sha:**
fc1836545 Merge pull request #4463 from ARMmbed/release-candidate

**DAPLink version:**

**Expected behavior**
Mbed cloud client should behave consistently regardless of when an OdinWiFiInterface is instantiated.

**Actual behavior**
Timers do not expire and thus Mbed cloud client does not send keep-alive messages in the case where an OdinWiFiInterface is instantiated after the cloud client.

**Steps to reproduce**
Apply easy-connect-patch.txt to the easy-connect library in mbed-cloud-client-example-restricted.  Execute mbed-cloud-client-example-restricted

----------------------------------------------------------------

0xc0170, 2017-09-27:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-10-03:
We have reproduced this and found the cause. The driver initializes the HAL ticks and configured the CPU clock but it has already been done by mbed OS/ST HAL. This will be fixed in the next release of the driver that most likely will be done this week.

0xc0170, 2017-10-05:
> This will be fixed in the next release of the driver that most likely will be done this week.

Great, waiting for the patch!

cmonr, 2018-06-28:
Closing since the referenced PR has long since been merged.

closed 2018-06-28 by cmonr