ARMmbed/mbed-os #5215: [UBLOX_EVK_ODIN_W2] PWM enabled only for LED2/3/D3/D5

andrewc-arm, 2017-09-28:
### Description
- Type: Question
- Priority: Minor

-----------------------------------------------------------------

Hi, I am testing UBLOX_EVK_ODIN_W2 due to Seoul IoT Hackathon planning (http://english.seoulhackathon.org/).
The engineers are questioning why PWM is enabled for LED2/LED3/D3/D5 only.
For other LED1 and other D* pins, PWM is not supported.

They can understand if only certain pins are PWM-enabled.
However, this limitation is not documented in the following EVK page.
https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/

Could we have some explanation about this?

Thanks,

0xc0170, 2017-09-28:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-09-28:
Hi, thanks for the feedback! It's correct that there is limited PWM on the EVK pins. It should be possible to also use A1 when configured as Digital via J26 as well as D2 and D8 if the USB UART is not used or reconfigured to use D0/D1(#5192).

We will add some comments on the EVK page or the user's guide.

andrewc-arm, 2017-09-29:
Thanks for the feedback!

closed 2017-09-29 by andrewc-arm