ARMmbed/mbed-os #5218: NUCLEO_F429ZI fails to continue from deepsleep

paviitan, 2017-09-28:
### Description

NUCLE0_F429ZI board does not wake up expectedly from deepsleep with user interrupt. 
 
Latest working mbed-os release: 5.5.1. 
Bug exists at least in mbed-os releases: 5.5.2 and 5.5.7. 

Known workaround: Leave only USE_PLL_HSI as clock source selection in targets.json  

- Type: Bug 

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI

**Toolchain:**
GCC_ARM 

**mbed-os (5.5.2) sha:**
8828635da

**Steps to reproduce:**
Run [Nucleo sleep example.](https://os.mbed.com/teams/ST/code/Nucleo_sleep/file/be046ad1730d/main.cpp) 
 
**Actual result:**
Program does not continue. 
 
**Expected result:**
Board continues program from deepsleep after user interrupt.

0xc0170, 2017-09-28:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-10-03:
I reproduced the problem. First of all this example has to be changed because the deepsleep function is now deprecated. Secondly I don't understand why we don't exit from the sleep function.

@0xc0170 It looks to me that it is related to the introduction of the sleep_manager. Any clue ?

bcostm, 2017-10-06:
I have replaced the mbed library by the mbed-os library in the example and the sleep/deepsleep works fine even if I still have the 'deprecated message' for the deepsleep function.

Is it normal that we can't use anymore sleep/deepsleep with mbed library ?

0xc0170, 2017-10-06:
It should have the same behaviour in this case (mbed-os or mbed lib). Be aware, sleep manager got in 5.6 so this issue is not related (reported for 5.5.x).

> Is it normal that we can't use anymore sleep/deepsleep with mbed library ?

All valid, since 5.6 however be aware that sleep is an entry function for sleep (that could be sleep or deepsleep, based on the sleep manager sleep counter state (deep sleep allowed od not). I'll have a closer look at the above example, and see if it is backward compatible.
cc @c1728p9

paviitan, 2017-11-30:
@0xc0170 Is it ok to go around the sleep manager with `hal_sleep` and `hal_deepsleep` ?
>I have replaced the mbed library by the mbed-oslibrary in the example and the sleep/deepsleep works fine 

@bcostm with 5.6.x I need to use `rtc_init();` at start of main block with my board if I start it "cold", after that the code seems to run as intended. Did you notice any similar issues?

jeromecoutant, 2017-11-30:
Hi

> I need to use rtc_init(); at start of main block with my board if I start it "cold", after that the code seems to run as intended. 

About RTC issue, is it possible to check if #5459 is changing board behavior as expected by you ?

Thx

paviitan, 2017-11-30:
@jeromecoutant #5459 seems to resolve any remaining problems I had with this example. Thanks!

bcostm, 2017-12-04:
Are you happy to close this issue if solved now ? Thanks.

ST_TO_BE_CLOSED

closed 2017-12-06 by paviitan