ARMmbed/mbed-os #5219: How do I get free_stack() of main thread?

mikevolzer-ep, 2017-09-28:
### Description
- Type: Question

---------------------------------------------------------------

## Question
How do I get the available stack of the main thread? I see there is a member function "free_stack", but I cannot find out how to get a reference to the main thread object

bulislaw, 2017-09-29:
Main thread is not a C++ object, maybe we should try to wrap it in some way. For now you'll need to use RTOS API directly http://arm-software.github.io/CMSIS_5/RTOS2/html/index.html
```
osThreadGetStackSpace(osThreadGetId());
```
executed from the main thread. You'll need to add that to your build command `-DMBED_STACK_STATS_ENABLED`.

closed 2017-09-29 by mikevolzer-ep

loverdeg-ep, 2017-09-29:
@bulislaw is there an option to have an entry point before the scheduler starts?

bulislaw, 2017-09-29:
Depending what you want to do, we have `mbed_sdk_init` but that's more for targets use, and then we have `mbed_main` that executes before `main`, but that's actually after the scheduler starts. Some more info https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/mbed_boot.c