ARMmbed/mbed-os #5221: new and delete operators should be handled by the Runtime memory tracing

javier-moreno-tridonic-com, 2017-09-29:
### Description
- Type: Enhancement

**Reason to enhance or problem with existing solution**

Due to the malloc wrapping done by those operators here:
https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.cpp

Some important information is destroyed as the traced "caller" will be always **new**, **new[]**... etc., preventing the identification of the class/function who really triggered the memory allocation. This complicates looking for leaks.

**Suggested enhancement**
 
I don't know if that would be a real option, but maybe those operators could be moved to https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_alloc_wrappers.cpp and integrated in the tracing mechanism.

0xc0170, 2017-09-29:
cc @c1728p9 @pan- @bulislaw

pan-, 2017-09-29:
That sounds like a reasonable approach to handle that issue.

loverdeg-ep, 2017-09-29:
Running into this same problem.
Temporarily workaround was providing a custom callback for trace, setting breakpoints logically, and analyzing the call stack.

0xc0170, 2018-06-22:
@TeroJaasko 5456 - this did not address this issue? What is missing to make this resolved?

0xc0170, 2018-07-03:
It should be resolved

closed 2018-07-03 by 0xc0170