ARMmbed/mbed-os #5222: u-blox EVK-ODIN-W2 switch PullUp resistance example

gggekov, 2017-09-29:
### Description
- Type: Enhancement 
- Target: ublox EVK-ODIN-W2

The SW0/SW1 switches need to be configured as pull-ups in the software application. It can take quite a lot of time to figure this out. 

## Enhancement

**Reason to enhance or problem with existing solution** Simplify the usage of user buttons on the ulbox EVK ODIN W2 board. 

**Suggested enhancement** Add the following example on the mbed page of the odin-w2 board.

```
#include "mbed.h"

int main(int, char**) 
{
  DigitalOut led(LED1);
  DigitalIn btn(SW0);
  btn.mode(PullUp);
  while (1) 
  {
        if( btn.read() )
            led = 1;
        else
            led = 0;

  }
}
```

0xc0170, 2017-09-29:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-09-29:
Thanks for the suggestion @gggekov We'll clarify it in the documentation.

0xc0170, 2017-10-02:
> Thanks for the suggestion @gggekov We'll clarify it in the documentation.

Let us know once done, to close this issue

andreaslarssonublox, 2017-10-17:
I have updated the board page now.

closed 2017-10-17 by gggekov