ARMmbed/mbed-os #5225: Make MBED_MEM_TRACING_ENABLED an application config option

loverdeg-ep, 2017-09-29:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
Enhancement
- Related issue: [Handbook-#275](https://github.com/ARMmbed/Handbook/issues/275)
- Priority: Minor

---------------------------------------------------------------

## Enhancement

**Reason to enhance or problem with existing solution**
Confusion regarding the difference between an application config option and target/toolchain specific options

**Suggested enhancement**
Make MBED_MEM_TRACING_ENABLED an application config option

**Pros**
Improved OS configuration intuitiveness

**Cons**
Unsure

0xc0170, 2017-10-02:
👍  for having this configurable via a config

cc @c1728p9

bmcdonnell-ionx, 2018-03-21:
For reference,

https://os.mbed.com/docs/v5.7/tutorials/optimizing.html

> ### Using the memory tracer
>
> The memory tracer is not enabled by default. To enable it, you need to define the `MBED_MEM_TRACING_ENABLED` macro. The recommended way to define this macro is to add it to the list of macros defined in your `mbed_app.json`:
>
> ```
> {
>     "macros": ["MBED_MEM_TRACING_ENABLED"]
> }
> ```

deepikabhavnani, 2018-12-28:
Resolved by : https://github.com/ARMmbed/mbed-os/pull/7402

closed 2018-12-28 by deepikabhavnani