ARMmbed/mbed-os #5241: mbed-os test build for REALTEK_RTL8195AM,IAR takes ~35 min

studavekar, 2017-10-02:
Command used to build 
```
mbed test --compile -m REALTEK_RTL8195AM -t IAR -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 --clean
```

### Description
- Type: Bug 
- Priority: Major 
---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
IAR

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
c60194f Merge pull request #5208 from jeromecoutant/PR_F7_RTC

**Expected behavior**
Build time should be 2-5 mins

**Actual behavior**
Build time is around ~35 mins

**Steps to reproduce**

mbed test --compile -m REALTEK_RTL8195AM -t IAR -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 --clean

0xc0170, 2017-10-04:
cc @tung7970 @Archcady

Archcady, 2017-10-04:
IAR elf2bin tool ( aka. ielftool ) is always trying to do zero-padding to the holes of address, which take a long time. Right now we have to cut out useful pieces of binary code from original bin file in post-build process. @0xc0170, any idea to stop ielftool from padding?

tung7970, 2017-10-04:
@Archcady  rom and image1 should be removed from ICF file for mbed. They are not actually referenced and causing huge zero-padding section.

0xc0170, 2017-10-04:
@tung7970 Please send a fix

tung7970, 2017-10-05:
After further check, the zero-padding issue is due to multiple separated memory regions. The zero-padding is causing lengthy process time and huge binary size. We were able to solve binary size issue with our post processing script, but not the zero-padding behavior of ielftool.

The reason for having multiple memory regions is to have fine-grain control over object placement. But I think it is not necessary for most cases. Let me check if I can come up with an easy solution.

tung7970, 2017-10-05:
@Archcady  Please help check if I have broken anything.

Archcady, 2017-10-05:
Thanks @tung7970 , It works.

0xc0170, 2017-11-02:
This is resolved via a referenced PR above

closed 2017-11-02 by 0xc0170