ARMmbed/mbed-os #5244: Compiling Odin Wifi driver fails with ARMC6 toolchain

JammuKekkonen, 2017-10-03:
### Description
- Type: Bug

Compiling Odin Wifi driver fails with ARMC6 toolchain

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
ARMC6

**mbed-os sha:**
cc7556a92

**Expected behavior**
Odin wifi driver compiles

**Actual behavior**
Error: L6218E: Undefined symbol typeinfo for NetworkInterface (referred from OdinWiFiInterface.o).
Finished: 0 information, 1225 warning and 1 error messages.

**Steps to reproduce**
try to compile as main.cpp:

#include "OdinWiFiInterface.h"
int main(void)
{
    OdinWiFiInterface wifi;
    wifi.connect("ssid", "pass", NSAPI_SECURITY_WPA_WPA2);
    return 0;
}

mbed compile -t ARMC6 -m UBLOX_EVK_ODIN_W2

JammuKekkonen, 2017-10-03:
@JanneKiiskila

0xc0170, 2017-10-04:
cc @andreaslarssonublox @andreaspeterssonublox The driver will need an update most probably for this new toolchain

andreaslarssonublox, 2017-10-04:
I tried to compile mbed-os(1.2.2) with ARMC6(6.5) with the sha version mentioned above and it failed to compile with the following reason:
        [DEBUG] Return: 8
        [DEBUG] Output: error: error opening 'C:\Users\ANDREA~1.d': permission denied
        [DEBUG] Output: 1 error generated.
        [DEBUG] Output: Fatal error: A1905U: Pre-processor step failed for 'C:\git\ARMmbed\mbed-os\rtos\TARGET_CORTEX\rtx5\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\irq_cm4f.S'
        [DEBUG] Output: 1 Error, 0 Warnings

Something you know how to fix @JammuKekkonen?

JanneKiiskila, 2017-10-04:
I think we've used a Linux machine for the building. Error opening though would imply an access issue, any reason why the compilation process would not be able to access your `c:\users\andrea~1.d` folder/file?

andreaslarssonublox, 2017-10-04:
I'm using windows 10. I can try it on a windows 7 machine as well. It works well with ARM 5.06u3 compiler though.

SeppoTakalo, 2017-10-09:
Might be related: https://github.com/ARMmbed/mbed-os/issues/5282

andreaslarssonublox, 2017-10-10:
Ok, should we close this then since it should not be possible to compile it according to #5282?
We can open a new PR for the "permission denied" issue if it persists when we add support for the ARMC6 compiler.

JanneKiiskila, 2017-10-10:
Yep, we should file a compiler bug for the ARMC 6 issue. 
Bigger question then is do you really to deliver binaries or could you deliver source code?
We'll have this same issue with every compiler change.

0xc0170, 2018-06-22:
@andreaslarssonublox @JanneKiiskila  there were few updates since this issue was discussed the last time. Is this issue still relevant?