ARMmbed/mbed-os #5248: Nucleo F429ZI LWIP6 cannot get IP (Connect fails)

teetak01, 2017-10-04:
### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI + LWIP6

**Toolchain:**
GCC_ARM

**mbed-os version:**
5.6.1

**Expected behavior**
Device gets IP

**Actual behavior**
Device never gets IP

[EasyConnect] MAC address 00:80:e1:2c:00:28
[EasyConnect] Connection to Network Failed -3010!
MainThread: Test Case fails because of: Failed to initialize connection

teetak01, 2017-10-04:
@adbridge  regression in Mbed OS 5.6.1

Timjun01, 2017-10-04:
Nucleo F429ZI cannot get ip address: all Syte test failing.

00:51:45 00:51:43.813 | D1 <-> MainThread: CMD ifconfig eth0 up, timeout=1507078353, time=1507078303
00:51:45 00:51:43.813 | D1 --> DutThread: ifconfig eth0 up
00:51:45 00:51:43.978 | D1 <-- DutThread: [EasyConnect] IPv4 mode
00:51:45 00:51:43.978 | D1 <-- DutThread: [EasyConnect] Using Ethernet
00:51:46 00:51:44.814 | D1     MainThread: Waiting for response... timeout=48
00:51:47 00:51:45.814 | D1     MainThread: Waiting for response... timeout=47
00:51:48 00:51:46.814 | D1     MainThread: Waiting for response... timeout=46
00:51:49 00:51:47.815 | D1     MainThread: Waiting for response... timeout=45
00:51:50 00:51:48.815 | D1     MainThread: Waiting for response... timeout=44
00:51:51 00:51:49.816 | D1     MainThread: Waiting for response... timeout=43
00:51:52 00:51:50.816 | D1     MainThread: Waiting for response... timeout=42
00:51:53 00:51:51.816 | D1     MainThread: Waiting for response... timeout=41
00:51:54 00:51:52.817 | D1     MainThread: Waiting for response... timeout=40
00:51:55 00:51:53.818 | D1     MainThread: Waiting for response... timeout=39
00:51:56 00:51:54.819 | D1     MainThread: Waiting for response... timeout=38
00:51:57 00:51:55.819 | D1     MainThread: Waiting for response... timeout=37
00:51:58 00:51:56.819 | D1     MainThread: Waiting for response... timeout=36
00:51:59 00:51:57.820 | D1     MainThread: Waiting for response... timeout=35
00:52:00 00:51:58.821 | D1     MainThread: Waiting for response... timeout=34
00:52:01 00:51:59.821 | D1     MainThread: Waiting for response... timeout=33
00:52:02 00:52:00.822 | D1     MainThread: Waiting for response... timeout=32
00:52:03 00:52:01.822 | D1     MainThread: Waiting for response... timeout=31
00:52:04 00:52:02.823 | D1     MainThread: Waiting for response... timeout=30
00:52:05 00:52:03.824 | D1     MainThread: Waiting for response... timeout=29
00:52:06 00:52:04.824 | D1     MainThread: Waiting for response... timeout=28
00:52:07 00:52:05.825 | D1     MainThread: Waiting for response... timeout=27
00:52:08 00:52:06.825 | D1     MainThread: Waiting for response... timeout=26
00:52:09 00:52:07.826 | D1     MainThread: Waiting for response... timeout=25
00:52:10 00:52:08.827 | D1     MainThread: Waiting for response... timeout=24
00:52:11 00:52:09.827 | D1     MainThread: Waiting for response... timeout=23
00:52:12 00:52:10.828 | D1     MainThread: Waiting for response... timeout=22
00:52:13 00:52:11.828 | D1     MainThread: Waiting for response... timeout=21
00:52:14 00:52:12.828 | D1     MainThread: Waiting for response... timeout=20
00:52:15 00:52:13.829 | D1     MainThread: Waiting for response... timeout=19
00:52:16 00:52:14.830 | D1     MainThread: Waiting for response... timeout=18
00:52:17 00:52:15.830 | D1     MainThread: Waiting for response... timeout=17
00:52:18 00:52:16.831 | D1     MainThread: Waiting for response... timeout=16
00:52:19 00:52:17.832 | D1     MainThread: Waiting for response... timeout=15
00:52:20 00:52:18.833 | D1     MainThread: Waiting for response... timeout=14
00:52:21 00:52:19.833 | D1     MainThread: Waiting for response... timeout=13
00:52:22 00:52:20.834 | D1     MainThread: Waiting for response... timeout=12
00:52:23 00:52:21.834 | D1     MainThread: Waiting for response... timeout=11
00:52:24 00:52:22.834 | D1     MainThread: Waiting for response... timeout=10
00:52:25 00:52:23.836 | D1     MainThread: Waiting for response... timeout=9
00:52:26 00:52:24.836 | D1     MainThread: Waiting for response... timeout=8
00:52:27 00:52:25.837 | D1     MainThread: Waiting for response... timeout=7
00:52:28 00:52:26.837 | D1     MainThread: Waiting for response... timeout=6
00:52:29 00:52:27.837 | D1     MainThread: Waiting for response... timeout=5
00:52:30 00:52:28.839 | D1     MainThread: Waiting for response... timeout=4
00:52:31 00:52:29.839 | D1     MainThread: Waiting for response... timeout=3
00:52:32 00:52:30.839 | D1     MainThread: Waiting for response... timeout=2
00:52:33 00:52:31.840 | D1     MainThread: Waiting for response... timeout=1
00:52:34 00:52:32.840 | D1     MainThread: Waiting for response... timeout=0
00:52:35 00:52:33.842 | D1     MainThread: CMD timeout: ifconfig eth0 up
00:52:35

Timjun01, 2017-10-04:
Wifi IPv4 work

0xc0170, 2017-10-04:
cc @bcostm @adustm @LMESTM @jeromecoutant

0xc0170, 2017-10-04:
I am looking at changeset for 5.6.1 - https://github.com/ARMmbed/mbed-os/pull/5184 (I can not spot there changes to this target neither lwip). I suspect 5.6.0 also have this problem, then we need to bisect more files to actually get the change that caused this

@adustm Can you help?

mikaleppanen, 2017-10-04:
I cannot reproduce this with our ipv6 connection (4g router) and F429ZI. Both example client and client testapp works ok with ipv6 ethernet.

JanneKiiskila, 2017-10-05:
Indeed, @mikaleppanen  is right - turns out we had a really strange CI networking issue - some boards didn't get an IP-address anymore, while some boards in the same setup did. We've never seen this kind of failure before and as some other boards were getting both IPv4 and IPv6 address we never guessed to suspect a network issue. But, that was the root cause for this incident. 

Apologies for the false alarm. Closing issue out.

teetak01, 2017-10-05:
Yeah, turned out be a test-lab issue. Closing. Nucleo is working again for us.

closed 2017-10-05 by teetak01