ARMmbed/mbed-os #5254: Wi-Fi WPA2-Enterprise support

loverdeg-ep, 2017-10-04:
## Question

What is the current status of  WPA2-Enterprise support for WifiInterface and any existing WifiInterface implementations?

Planning?, progressing but private?, not on the roadmap? etc..

I'm seeing a couple projects in which stakeholders want to make WPA2-Enterprise a requirement.

0xc0170, 2017-10-05:
cc @SeppoTakalo @sg-

lauri-piikivi, 2017-10-06:
Enterprise WPA support is on our backlog, but has not gained enough demand to make it to the roadmap yet. It would be nice to understand better the your needs for this, which mode and in what schedule? If that can not be written here, please contact me via mail lauri dot piikivi at arm dot com

loverdeg-ep, 2018-01-04:
@0xc0170 @lauri-piikivi 
This is a major blocking issue for us. 
All of our current industrial/enterprise leads have wpa2-enterprise as a requirement.

Low cost and low power embedded WiFi modules are going to play a large part in the IoT endpoint space.

I don't think WPA2 is going to cut it for big installations in which the endpoints will be batch provisioned to a Wi-Fi network in the factory or by the end customer; Nor where each device on the network needs it's own credentials.

maansaari, 2018-01-08:
Hi,

At this moment we haven't received enough customer feedback/requests for WPA2-enterprise support in mbed OS wifi.  At the moment WPA-enterprise is quite fragmented with different flavours of EAP-TLS, which makes the requirement quite unclear at this moment. 
In case you are interested to discuss the requirement in more details, please contact me via email: kirsi.maansaari@arm.com 

Best Regards, Kirsi
Kirsi Maansaari
mbed OS Connectivity Product Manager

0xc0170, 2018-06-22:
@loverdeg-ep I believe this question is answered by @maansaari .

I'll close it.

closed 2018-06-22 by 0xc0170