ARMmbed/mbed-os #5259: Override CLOCK_SOURCE via mbed_app.json ?

janjongboom, 2017-10-05:
### Description
- Question

## Question

As asked [here](https://os.mbed.com/questions/79176/NUCLEO-F302R8-with-HSI/). Is it possible to override CLOCK_SOURCE through mbed_app.json ?

0xc0170, 2017-10-05:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-10-05:
Hi
Yes, it is possible to override CLOCK_SOURCE.

Here is an example of mbed_app.json:

{
    "target_overrides": {
        "NUCLEO_F410RB": {
            "target.clock_source": "USE_PLL_HSI"
        }
    }
}

I agree this is not so easy to use and understand... Maybe ARM should update MBED page:
https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/config_system/

Have also a look on ST wiki page:
https://os.mbed.com/teams/ST/wiki/Automatic-clock-configuration

janjongboom, 2017-10-05:
Thanks @jeromecoutant.

@iriark01 I see we don't have a section for target settings in https://os.mbed.com/docs/v5.6/reference/configuration-overview.html, would be good to add.

iriark01, 2017-10-05:
@AnotherButler

AnotherButler, 2017-10-05:
@janjongboom You can find the section about adding and configuring targets here: https://os.mbed.com/docs/v5.6/tools/adding-and-configuring-targets.html

janjongboom, 2017-10-06:
@AnotherButler Nothing there mentions the ability to override target properties through mbed_app.

theotherjimmy, 2017-10-06:
@janjongboom  see https://os.mbed.com/docs/v5.6/tools/the-configuration-system.html#overriding-target-attributes

janjongboom, 2017-10-09:
@theotherjimmy ok, cool.

closed 2017-10-09 by janjongboom