ARMmbed/mbed-os #5265: what is importance of DEVICE_I2C_ASYNCH and strategy here?

peeyushpashine, 2017-10-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Related issue: DEVICE_I2C_ASYNCH
- Priority: Minor

---------------------------------------------------------------
## Question

What is the significance of DEVICE_I2C_ASYNCH flag for the targets? I see it was added for Nucleo F031K6 recently. @bcostm

0xc0170, 2017-10-06:
It was added as an addition to regular i2c API that provides blocking API - async API (write/read with event callbacks).

bcostm, 2017-10-06:
With the asynchronous mode, you can send a transfer and do something else in the meantime (process more code or go to sleep). You will receive a notification via a callback when the transfer is finished.

closed 2017-10-23 by peeyushpashine