ARMmbed/mbed-os #5266: Build errors in online compiler for lpc812, lpc1549, lpc824, ssci824

janjongboom, 2017-10-06:
Error: The build system did not finish successfully.
Info: Unable to download. Fix the reported errors...

This is happend if I use following target:
NXP LPC800-MAX (lpc812)
LPCXpresso1549 (lpc1549)
LPCXpresso824-MAX (lpc824)
Switch Science mbed LPC824 (ssci824)

Not depends on library version or web browser.  No additional message is shown by enabling the Verbose check-box (it does not seem to kick compiler).  Few developer in Japan contacted me about this issue today.

Can someone find out this?

Also reported here https://os.mbed.com/questions/79189/I-got-Compiler-Error-230-from-yesterday-/

@theotherjimmy @screamerbg

toyowata, 2017-10-12:
Building with Mbed CLI for these targets should still work.

https://os.mbed.com/users/MACRUM/notebook/how-to-build-micro-library-target-by-mbed-cli
https://os.mbed.com/users/MACRUM/notebook/using-mbed-cli-with-LPC824/

janjongboom, 2017-10-12:
cc @theotherjimmy @screamerbg @sg-

neilt6, 2017-10-13:
I'm experiencing this issue as well on both the pre-compiled mbed library, and mbed-dev.

toyowata, 2017-10-24:
@theotherjimmy 
Now, I got error below:
```
Error: Cannot open source input file "cmsis_armcc.h": No such file or directory in "extras/mbed_235179ab3f27/TARGET_LPC824/cmsis_compiler.h", Line: 34, Col: 28
```

theotherjimmy, 2017-10-24:
@toyowata Confirmed with `mbed_blinky` using the most recent version of `mbed` and an LPC824. I'll have a deeper look today.

theotherjimmy, 2017-10-24:
Ah, it looks like there is no ARM build for these targets in 5.6.3's mbed 2 release. There will be one in 5.6.4, so we'll see that tomorrow.

toyowata, 2017-11-10:
Fixed in mbed library v157.

PR #5317 (2017-10-23): Tools: Check for toolchain and core support for Arm Compilers
Fixes #5266 when deployed to the online compiler
Fixes #5345

Also corrects offline compiling of uARM for every target that supports uARM but
not ARM.

TODO:
 - [ ] A test so that I can't break this again
closed 2017-11-10 by toyowata