ARMmbed/mbed-os #5267: CAN read doesn't work when the interrupt is attached.

lightway29, 2017-10-06:
### Description
- Type: Bug 
- Priority: Major

This is a simple CAN read sample.Which is suppose to flip the led1 when a can message is received.The attached interrupt function is never called even though the messages received.

_Note - CAN messages are sent from another device with an interval of 1sec._

```c++
#include "mbed.h"

DigitalOut led1(LED1);
CAN can(p9, p10);
CANMessage msg;

bool msg_rx_flag = 0;

void can_interrupt() {
  can.attach(NULL);
  msg_rx_flag = 1;
}

int main() {
  printf("[DEBUG LOG] Main loop started. \n\r");

  can.mode(CAN::Normal);
  can.frequency(100000);  // Initialise mbed CAN with a baudrate of 100 kbps
  can.attach(&can_interrupt, CAN::RxIrq);

  while (1) {
    if (msg_rx_flag == 1) {
      if (can.read(msg)) {
        led1 = !led1;
      }
      msg_rx_flag = 0;
      can.attach(&can_interrupt, CAN::RxIrq);
    }
  }
}

```
But this works without any issue
```c++
#include "mbed.h"

DigitalOut led1(LED1);
CAN can(p9, p10);
CANMessage msg;

bool msg_rx_flag = 0;

void can_interrupt() {
  if (can.read(msg)) {
    led1 = !led1;
  }
}

int main() {
  printf("[DEBUG LOG] Main loop started. \n\r");

  can.mode(CAN::Normal);
  can.frequency(100000);  // Initialise mbed CAN with a baudrate of 100 kbps
  can.attach(&can_interrupt, CAN::RxIrq);

  while (1) {
  }
}

```

---------------------------------------------------------------
## Bug

**Target**
NXP LPC1768

**Toolchain:**
GCC_ARM

**Toolchain version:**
1.60301.0

**mbed-os(5.5.6) sha:**
mbed_lib_rev152 

**Expected behavior**
The registered interrupt should be triggered when a message is received.And the led1 should be flipped.
**Actual behavior**
The interrupt function is not trigger when simple flag is set to be true and read can messages in the main loop like in the mbed sample.
**Steps to reproduce**

1. Setup a device to send can messages(Extended) with a delay of one second.
2.Run the first code sample.
3.Run the second code sample

0xc0170, 2017-10-06:
`bool msg_rx_flag = 0;` this should be defined with qualifier `volatile`

cc @toyowata

lightway29, 2017-10-09:
@0xc0170  Updated the code as follow.
cc @vargham
- Related issue: `https://github.com/ARMmbed/mbed-os/issues/3527#issue-198794877`

```C++
#include "mbed.h"

DigitalOut led1(LED1);
CAN can(p9, p10);
CANMessage msg;
volatile bool msg_rx_flag = 0;

void can_interrupt() {
  can.attach(NULL);
  msg_rx_flag = 1;
}

int main() {
  printf("[DEBUG LOG] Main loop started. \n\r");

  can.mode(CAN::Normal);
  can.frequency(100000);  // Initialise mbed CAN with a baudrate of 100 kbps
  can.attach(callback(&can_interrupt), CAN::RxIrq);

  while (1) {
    if (msg_rx_flag == 1) {
      if (can.read(msg) == 1) {
        printf(
            "Received CAN message | id: %x, data: [0] %x, [1] %x, "
            "[2] %x, [3] %x, [4] %x, [5] %x, [6] %x, [7] %x \r\n",
            msg.id, msg.data[0], msg.data[1], msg.data[2], msg.data[3], msg.data[4], msg.data[5], msg.data[6],
            msg.data[7]);
      }
      led1 = !led1;
      msg_rx_flag = 0;
      can.attach(callback(&can_interrupt), CAN::RxIrq);
    }
  }
}
```

According to the updated code the interrupts are fired properly and the "msg_rx_flag" is set and cleared as expected but, the can.read returns '0' always unless its called in the ISR context.

-----------------------------------------------------------------

## Question

**Why won't the CAN read() function work in the user context ?**

cmonr, 2019-01-30:
@ARMmbed/team-nxp @ARMmbed/mbed-os-hal

devangel77b, 2019-07-23:
From the OP, if I try the first code it fails, but the second code that calls CAN.read() within the ISR also fails because of a Mutex. I am trying to use CAN.attach to handle CAN message receipt in OS 5.

allenman85, 2020-11-02:
I still see this issue in MBED 5.15.  Was it ever closed?

0xc0170, 2020-11-03:
> I still see this issue in MBED 5.15. Was it ever closed?

Unfortunately it has not yet.