ARMmbed/mbed-os #5276: Export fails when passing `-c`

LMESTM, 2017-10-09:
As first reported here: https://github.com/ARMmbed/mbed-os-example-blinky/issues/86
I'm facing an issue that seems to be related to MBED OS tools as reported by @0xc0170 

## Description

I imported this example in my environment 
`mbed` import `mbed-os-example-blinky`

Compiled ok:
` mbed compile -t IAR -m NUCLEO_L476RG `

Trying export fails:
```
$ mbed export -i IAR -m NUCLEO_L476RG  -c -vv
Traceback (most recent call last):
  File "C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py", line 256, in <module>
    main()
  File "C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py", line 248, in main
    rmtree(BUILD_DIR)
  File "c:\python27\lib\shutil.py", line 239, in rmtree
    onerror(os.listdir, path, sys.exc_info())
  File "c:\python27\lib\shutil.py", line 237, in rmtree
    names = os.listdir(path)
WindowsError: [Error 3] The system cannot find the path specified: 'C:\\Data\\Workspace\\mbed-os-example-blinky\\mbed-os\\BUILD/*.*'
[mbed] Working path "C:\Data\Workspace\mbed-os-example-blinky" (program)
[mbed] Exec "c:\python27\python.exe -u C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m NUCLEO_L476RG -c --source ." in C:\Data\Workspace\mbed-os-example-blinky
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m NUCLEO_L476RG -c --source ." in "C:\Data\Workspace\mbed-os-example-blinky"
---
```

Windows OS, with MINGW64

0xc0170, 2017-10-09:
cc @theotherjimmy

theotherjimmy, 2017-10-09:
@LMESTM @0xc0170 Yeah. I'll remove `-c`. It no longer makes sense.

closed 2018-03-28 by ChiefBureaucraticOfficer