ARMmbed/mbed-os #5279: Ticker interrupt executed periodically

mprse, 2017-10-09:
### Description
- Type: Bug
- Priority: Minor

While working with:
- PR https://github.com/ARMmbed/mbed-os/pull/5233 
- PR https://github.com/ARMmbed/mbed-os/pull/5234
I noticed that ticker interrupt is handled in different way on K64F platform.
I have checked the following platforms: K64F, NUCLEO_F429ZI, NUCLEO_F070RB, NUCLEO_L476RG, NRF51_DK, ARCH_PRO.

---------------------------------------------------------------
## Bug

**Target**
- K64F

**Toolchain:**
- GCC_ARM
- ARM
- IAR

**mbed-cli version:** 
1.2.2

**Expected behavior**
us_ticker_set_interrupt() function shall set point in time when ticker interrupt will be triggered. When this point of time is reached ticker interrupt is triggered once.

**Actual behavior**
On K64F board interrupt is triggered periodically. 
If we call us_ticker_set_interrupt(us_ticker_read() + INT_TIME), then interrupt will be triggered in the following points of time:
- read time + INT_TIME,
- read time + 2 *INT_TIME,
- read time + 3 *INT_TIME,
etc.

**Steps to reproduce**

The following code can be used to reproduce the issue:
```
uint32_t timestamp[10] = { 0 };
uint32_t cnt = 0;
void ticker_event_handler_stub(const ticker_data_t * const ticker) {
    us_ticker_clear_interrupt();
	if (cnt < 10) {
		timestamp[cnt] = us_ticker_read();
		cnt++;
	}
}

void test(void) {
	set_us_ticker_irq_handler(ticker_event_handler_stub);
	us_ticker_init();
	const uint32_t now = us_ticker_read();
	us_ticker_set_interrupt(now + 1000); // 1 ms
	wait_ms(10);
	us_ticker_disable_interrupt();
	for (int i = 0; i < 10; i++) {
		printf("irq timestamp %d: %d. \n", i, timestamp[i]);
	}
}
```

This code gives the following results:

> NUCLEO_F070RB
> irq timestamp 0: 1623120.
> irq timestamp 1: 0.
> irq timestamp 2: 0.
> irq timestamp 3: 0.
> irq timestamp 4: 0.
> irq timestamp 5: 0.
> irq timestamp 6: 0.
> irq timestamp 7: 0.
> irq timestamp 8: 0.
> irq timestamp 9: 0.
> 
> K64F:
> irq timestamp 0: 1419010.
> irq timestamp 1: 1420011.
> irq timestamp 2: 1421012.
> irq timestamp 3: 1422013.
> irq timestamp 4: 1423014.
> irq timestamp 5: 1424015.
> irq timestamp 6: 1425016.
> irq timestamp 7: 1426017.
> irq timestamp 8: 1427018.
> irq timestamp 9: 0.

On NUCLEO_F070RB board IRQ handler is executed only once.
On K64F IRQ handler is executed every 1 ms period.

Note:
To reproduce the issue API to set ticker interrupt handler is required provided in PR https://github.com/ARMmbed/mbed-os/pull/5234.

0xc0170, 2017-10-09:
cc @mmahadevan108

mmahadevan108, 2017-10-13:
PR https://github.com/ARMmbed/mbed-os/pull/5316 should fix this issue.

mprse, 2017-10-17:
Provided fix resolves this issue.

mprse, 2018-05-08:
This issue can be closed.

closed 2018-05-08 by 0xc0170