ARMmbed/mbed-os #5280: FlashIAP Error Handling

Alonof, 2017-10-09:
### Description
- Type:  Enhancement 
- Priority:  Minor

## Enhancement

All FlashIAP APIs return "-1" has error, there is no way of knowing what is the real error

**Suggested enhancement**
Add Error opcodes to describe each failure 

**Pros**

Good for debugging

**Cons**

No Cons

0xc0170, 2017-10-09:
This will require error codes definition for flash HAL API, to extend the current return values.

cc @bulislaw @c1728p9 @pan-

deepikabhavnani, 2017-11-29:
Internal Tracking: https://jira.arm.com/browse/IOTMORF-1775

JammuKekkonen, 2019-02-22:
FlashIAP::init will also leak memory if called multiple times

0xc0170, 2019-02-22:
> FlashIAP::init will also leak memory if called multiple times

I would assume this is separate issue and can be fixed? Please send a patch!

SeppoTakalo, 2020-02-26:
This is a proposal for Mbed OS HAL API.
I think we should not do it as it pushed an error handling that should be done in the driver, up to the application.
If the underlying flash driver needs retrying, it should be done in the driver level, not in the application.
So I close the ticket, as the `-1` is enough for now, and we should instead fix the failing drivers, instead of pushing the retry logic to application.

closed 2020-02-26 by SeppoTakalo