ARMmbed/mbed-os #5281: FlashIAP::program failed on ODIN board

Alonof, 2017-10-09:
### Description
- Type: Bug
- Priority:  Major

## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
1.2.0

**Expected behavior**
FlashIAP:: program failed programming the internal flash on Odin Board.
Same code that runs on NUCLEO F429ZI ( the same CPU has ODIN) passes while in ODIN is failing 

**Actual behavior**
calling FlashIAP:: program on ODIN

**Steps to reproduce**

0xc0170, 2017-10-09:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-10-10:
Hi @Alonof,
We have tested it for the latest driver release candidate here https://github.com/ARMmbed/mbed-os/pull/5287 and also for the previous https://github.com/ARMmbed/mbed-os/pull/4567 and the FlashIAP test cases passed.

Can you provide the following:
- test application
- mbed-os commit SHA
- toolchain versions

0xc0170, 2017-10-13:
@Alonof bump

Alonof, 2017-10-16:
after running the tests today 
there is no failure on the flash API

Alonof, 2017-12-28:
reproduce steps

deploy and compile PAL
https://github.com/ARMmbed/pal-example

Run PAL example bin
Flash is failing with the following errors:

Programming alignment error
Programming parallelism error

Alonof, 2017-12-28:
https://jira.arm.com/browse/IOTPART-3659

0xc0170, 2018-01-08:
@Alonof The repository is private, can you share public one to be able to test it?

Looking at the errors, they both come from ST HAL, wrong usage of their driver? the flash implementation is shared for all STM32F4 targets

Alonof, 2018-01-28:
Hi @0xc0170 I See you have permission for PAL repository 
These errors come from the H/W,
The same code is running on NUCLEO and ODIN boards but only on ODIN it fails

0xc0170, 2018-01-29:
> Hi @0xc0170 I See you have permission for PAL repository

It was not about me, but a bug report should have all available information publicly accessible.
As @andreaslarssonublox requested above https://github.com/ARMmbed/mbed-os/issues/5281#issuecomment-335427932 , to be able to reproduce them, they will need those information.

> FlashIAP:: program failed programming the internal flash on Odin Board.

What is the exact failure, error return code? Any debug info ? Call stack?

Tests might be passing but app failing (I would like to find out if we are missing test coverage or where is this bug coming from).

Alonof, 2018-01-30:
Hi 
The errors come from the flash register.
here are the errors:
1. Programming alignment error
2. Programming parallelism error

Error 1 - the driver should handle the writing.
Error 2 - Also the driver should handle parallelism flow, also the flash APIs are blocking

When running PAL flow the flash is failing with the above erros

0xc0170, 2018-01-30:
For those 2, what are the input parameters (address, sizes of data - to know if multiple banks are involved), if we can reproduce  the arguments in our test, or at least small code snippet.

andreaslarssonublox, 2018-02-01:
No, I don't recognize the errors and need the test application to be able to reproduce it. If this can't be shared publicly you can send it to my e-mail instead.

0xc0170, 2018-06-20:
> No, I don't recognize the errors and need the test application to be able to reproduce it. If this can't be shared publicly you can send it to my e-mail instead.

@Alonof please provide the small example to reproduce the issue. Previously private repository was referenced

0xc0170, 2018-08-02:
Can't reproduce, reopen with an update

closed 2018-08-02 by 0xc0170