ARMmbed/mbed-os #5282: Compiling for ARMC6 picks up TOOLCHAIN_ARM folders also

SeppoTakalo, 2017-10-09:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Toolchain:**
ARMC6

**Expected behavior**
When compiling Mbed OS, it should pick up folder `TOOLCHAIN_X` only if the `X` matches the toolchain you are using.

**Actual behavior**
When compiling using `ARMC6` it picks up also `TOOLCHAIN_ARM` folders.

**Steps to reproduce**
```
mbed import mbed-os-example-mesh-minimal
cd mbed-os-example-mesh-minimal
mbed target K64F
mbed toolchain ARMC6
mbed compile
```

Compilation should fail because we have not provided any binaries for ARMC6 toolchain. See [here](https://github.com/ARMmbed/mbed-os/tree/master/features/nanostack/FEATURE_LOWPAN_ROUTER)

Compilation how ever succeed. This means that it was using binary blobs targeted for Arm compiler 5.

Arm compiler 5 and 6 are entirely different compilers. First one is developed within ARM. The version 6 is based on LLVM/Clang.
Event that these both use the same object formats, we cannot guarantee the compatibility.
It may work, but it might also work when linking against GCC objects.

0xc0170, 2017-10-09:
cc @theotherjimmy

SeppoTakalo, 2017-10-11:
Related to this, when compiling Nanostack border router using ARMC6 it asserts when initialising the RF driver.

```
[INFO][app ]: Starting NanoStack Border Router...
--
[INFO][app ]: Build date: Oct  2 2017 10:21:23
[INFO][brro]: NET_IPV6_BOOTSTRAP_AUTONOMOUS
[INFO][app ]: Using ETH backhaul driver...
[DBG ][brro]: Backhaul driver ID: 1
[DBG ][brro]: Create Mesh Interface
[INFO][brro]: thread_if_id: -1
mbed assertation failed: thread_if_id>=0, file: ./source/borderrouter_thread_tasklet.c, line 330
 ```

theotherjimmy, 2017-10-13:
> When compiling Mbed OS, it should pick up folder TOOLCHAIN_X only if the X matches the toolchain you are using.

This is not how toolchains work. `TOOLCHAIN_ARM_STD` and `TOOLCHAIN_ARM_MICRO` may also be present with `TOOLCHAIN_ARM`.

TL;DR: This is expected behavoir. Use `TOOLCHAIN_ARM_STD` for armc5-only stuff.

SeppoTakalo, 2017-10-17:
@theotherjimmy 
Please link me to the document that explains these?

We need to provide binaries for all of the supported toolchains and therefore need to separate between all versions of ARM compilers.

I have assumed following:

|Toolchain|Folder|
|----------|------|
|ARM compiler 5|`TOOLCHAIN_ARM`|
|ARM compiler 6| `TOOLCHAIN_ARM6` |
|GCC|`TOOLCHAIN_GCC`|
|IAR|`TOOLCHAIN_IAR`|

If the above is not true, can you provide me exact values.

For the reference, please look how [Nanostack binaries are provided](https://github.com/ARMmbed/mbed-os/tree/master/features/nanostack/FEATURE_THREAD_BORDER_ROUTER).
It matches how for example [Ublox drivers provide their binaries](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_UBLOX_EVK_ODIN_W2/sdk).

Now that we know that ARMc5 and ARM6 are not 100% compatible, we need this documented because it affects platform drivers also.

SeppoTakalo, 2017-10-17:
TL;DR If `TOOLCHAIN_ARM_STD` is for ARM c5 only, then what is for ARM6 only?

0xc0170, 2017-10-17:
I would have assumed the same as above table (ARM for ARMCC 5, ARM6 for ARMCC 6). As I build with -t ARM/ARM6. 
If I am creating a file like a startup file that is different for all 3, I would have a structure in the target tree like:

```
- TOOLCHAIN_ARM_STD
- TOOLCHAIN_ARM_MICRO
- TOOLCHAIN_ARM6
```

😕  But this is again we haven't yet unified STD/MICRO startups (I believe they could be). And its not toolchain, just a attribute of toolchain.

theotherjimmy, 2017-10-17:
@SeppoTakalo @0xc0170 The compiler is called ARM *C* 6. Please use the full name, otherwise you will confuse anyone else that reads this discussion.

So, corrections:
from @0xc0170:
As I build with -t ARM/ARMC6.

If I am creating a file like a startup file that is different for all 3, I would have a structure in the target tree like:
- `TOOLCHAIN_ARM_STD`
- `TOOLCHAIN_ARM_MICRO`
- `TOOLCHAIN_ARMC6`

>😕 But this is again we haven't yet unified STD/MICRO startups (I believe they could be). And its not toolchain, just a attribute of toolchain.

That still does not sit right with me, but I don't know of a good way to get rid of it just yet.

SeppoTakalo, 2017-10-18:
Sorry about naming confusion, I'll update my table.

So basically we have assumed following:

|Compiler|`mbed target <name>` |Target folder name|
|--------|-------------|------------------|
|ARM compiler 5|`ARM`|`TOOLCHAIN_ARM`|
|ARM compiler 6|`ARMC6`| `TOOLCHAIN_ARMC6` |
|GNU ARM Embedded|`GCC_ARM`|`TOOLCHAIN_GCC`|
|IAR Embedded 7.x or 8.x|`IAR`|`TOOLCHAIN_IAR`|

If the above is not true, it should be documented that what are the equivalent folder names. Otherwise 100% binary compatibility cannot be promised.

theotherjimmy, 2017-10-18:
> If the above is not true,

The above is not true

For starters, `mbed target <name>` would try to compile for the MCU named `ARM` for example. 
The corrected table follows:

|Compiler|`mbed toolchain <name>` |Exclusive folder name|
|--------|-------------|------------------|
|ARM compiler 5|`ARM`|`TOOLCHAIN_ARM_STD`|
|ARM compiler 5|`uARM`|`TOOLCHAIN_ARM_MICRO`|
|ARM compiler 6|`ARMC6`| `TOOLCHAIN_ARMC6` |
|GNU ARM Embedded|`GCC_ARM`|`TOOLCHAIN_GCC`|
|IAR Embedded 7.x or 8.x|`IAR`|`TOOLCHAIN_IAR`|

SeppoTakalo, 2017-10-19:
Thanks for clarifying. I meant `toolchain` instead of `target` of course.

So yes, yes, this bug report is valid. When compiling for ARMC6 it picks up the TOOLCHAIN_ARM and TOOLCHAIN_ARM_STD folders because [it inherits](https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L273) ARM_STD.

Also the table above, what you posted, is not properly documented.

theotherjimmy, 2017-10-19:
@SeppoTakalo You're right. I have updated my comment to include a "None" in that column.

theotherjimmy, 2017-10-19:
In general, we want to use `TOOLCHAIN_ARM`, `TOOLCHAIN_GCC` and `TOOLCHAIN_IAR`. I'm not going to stop someone outside of mbed-os from using `TOOLCHAIN_ARMC6` and `TOOLCHAIN_ARM_MICRO`, but I don't want to encourage it. That's why it's not documented behavior: I want to be able to change that.

artokin, 2017-10-25:
At the moment our applications fails to start when compiled with ARMC6 (together with nanostack binaries). We can not solve this until mbed-os build system is fixed. Therefore this fix is critical for us.

We can create a PR to mbed-os where nanostack binaries are located in folders like @SeppoTakalo mentioned above (ARM in TOOLCHAIN_ARM, ARMC6 in folder TOOLCHAIN_ARMC6, etc...). Is this right approach to proceed? Would mbed-os build system then select binaries based on used toolchain?

Who is going to update mbed-os build system so that correct binaries are selected?

artokin, 2017-10-25:
PR https://github.com/ARMmbed/mbed-os/pull/5381 contains nanostack ARMC6 binaries for testing.

sg-, 2017-10-25:
There should be a branch for ARMC6 where TOOLCHAIN_ARM contains only ARMC6 builds. That branch will be tested with ARMC6 and merged to master when all updates are made and testing passes.

theotherjimmy, 2017-10-25:
@sg- That branch will be `feature-armc6` on this repo. I intend to collect all of the required changes to update`-t ARM` to be Arm Compiler 6 and submit that entire branch as a PR in < 5 weeks time.

artokin, 2017-10-26:
This approach means that ARM compiler 5 will be discontinued in mbed OS 5.7. Is it really intended?

sg-, 2017-10-28:
Yes, we're only intending to support a single range of versions for any of the 3 compilers.

maansaari, 2017-10-30:
As this was discussed in Product management it was clear that we only change the default compiler to ARMC6, but the older one has to be supported too. So there has to be option to use ARMC5 for some time, before it can be deprecated. So for nanostack we need to provide both binaries.

SeppoTakalo, 2017-10-30:
With the chosen approach, this is not technically possible.

`TOOLCHAIN_ARM` can contain either version 5 or 6 binaries. If this bug is not fixed, then we only support version 6.

JanneKiiskila, 2018-01-26:
Can we get this one moving, please?

theotherjimmy, 2018-01-26:
Hi @JanneKiiskila,

I have no plans to change the current state of the scanning rules.

cmonr, 2018-06-26:
Making a note here. This is still an issue, but there is ongoing work to sort this out.

theotherjimmy, 2018-10-23:
@cmonr I'm not aware of any ongoing work to sort this out. Closing as I hove not plans to change this behavior.

closed 2018-10-23 by theotherjimmy