ARMmbed/mbed-os #5290: Cannot build NUCLEO_L011K4 with uARM anymore

bcostm, 2017-10-10:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L011K4

**Toolchain:**
uARM

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
df88a9dcc

**Expected behavior**
Build OK

**Actual behavior**
```
Traceback (most recent call last):
  File "./tools/build.py", line 212, in <module>
    build_profile=profile)
  File "C:\github\mbed-os\tools\build_api.py", line 1001, in build_mbed_libs
    build_profile=build_profile, jobs=jobs, clean=clean)
  File "C:\github\mbed-os\tools\build_api.py", line 358, in prepare_toolchain
    extra_verbose=extra_verbose, build_profile=profile)
  File "C:\github\mbed-os\tools\toolchains\arm.py", line 53, in __init__
    raise NotSupportedException("ARM compiler support is required for ARM build")
NotSupportedException: ARM compiler support is required for ARM build
```

**Steps to reproduce**
python ./tools/build.py -v -x -j 4 -m NUCLEO_L011K4 -t uARM

0xc0170, 2017-10-11:
cc @theotherjimmy

0xc0170, 2017-10-11:
I can confirm this, reproduced locally. Seems like ARM Micro object does not overwrite this exception (it only checks for ARM keyword, for uARM it should be uARM instead of ARM)

closed 2017-12-16 by sg-