ARMmbed/mbed-os #5294: EthernetInterface static IP socket connect error

berzin11295, 2017-10-11:
## Bug
Error -3004 is occured when I'm trying to establish socket connection. If I'm using dhcp, it works, but sometimes (randomly) it can't get ip address and returns -3009.

Board NUCLEO F767ZI
MBED-OS 5.6
```

#include "mbed.h"
#include "TCPSocket.h"
#include "EthernetInterface.h"
 
EthernetInterface eth;
TCPSocket socket;
 
#define IP         "192.168.1.177"
#define GATEWAY    "192.168.1.1"
#define MASK       "255.255.255.0"
 
int main()
{
    printf("Example network-socket TCP Server\r\n");
    eth.disconnect();
    int i=eth.set_network(IP,MASK,GATEWAY);
    printf("set IP status: %i \r\n",i);
    i=eth.connect();
    printf("connect status: %i \r\n",i);
    const char *ip = eth.get_ip_address();
    const char *mac = eth.get_mac_address();
    printf("IP address is: %s\n\r", ip ? ip : "No IP");
    printf("MAC address is: %s\n\r", mac ? mac : "No MAC");

    while(true) {
        socket.close();
        socket.open(&eth);
 
        socket.connect("93.175.29.245", 80);
 
        char sbuffer[] = "GET /checkPin.php?\r\n";
        int scount = socket.send(sbuffer, sizeof sbuffer);
 
        char rbuffer[64];
        int rcount = socket.recv(rbuffer, sizeof rbuffer);
        printf("recv %d [%.*s]\r\n", rcount, strstr(rbuffer, "\r\n")-rbuffer, rbuffer);
    }
}
```
Thank you for your attention!

0xc0170, 2017-10-11:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-10-12:
Just to inform others that I have tested the above code on the NUCLEO_F767ZI and I have reproduced the issue (error -3004).

I have also tested the same code on a NUCLEO_F746ZG and NUCLEO_F429ZI and no problem seen.

LMESTM, 2017-10-31:
@berzin11295 
would you be able to test workaround proposed in #5411 ? 
I have tested it ok on my side. I will run more non reg testing locally.

berzin11295, 2017-11-03:
Yes, it works! Thank you:)
(IAR 8.2, I've just changed file stm32xx_emac.c)

closed 2017-11-03 by berzin11295

LMESTM, 2017-11-03:
Great ! Thanks for the feedback