ARMmbed/mbed-os #5298: LPC1768: Ethernet

chrissnow, 2017-10-11:
### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
ARM

**Expected behavior**
TCPSocket_Example should work

**Actual behavior**
```
IP address is: No IP
sent -3005 [GET / HTTP/1.1]
recv -3005 []
Done
```

**Steps to reproduce**
Build the TCPSocket_Example with 5.5.7 and it will be fine, any 5.6\master will fail to get an IP, using static IP will still fail to work properly.

@geky I think you have helped me with similar before. any thoughts?

0xc0170, 2017-10-11:
cc @kjbracey-arm @SeppoTakalo

toyowata, 2017-10-13:
@chrissnow 

I didn't get any problem with mbed-os-example-sockets example.

[Result]
```
Ethernet socket example
IP address: 192.168.0.8
Netmask: 255.255.255.0
Gateway: 192.168.0.1
sent 39 [GET / HTTP/1.1]
recv 205 [HTTP/1.1 200 OK]
External IP address: 217.140.104.200
Done
```

[Build steps]

```
$ mbed import https://github.com/armmbed/mbed-os-example-sockets
$ cd mbed-os-example-sockets
$ mbed compile -m LPC1768 -t GCC_ARM -f
```

mbed-os-example-sockets (b32ff45174cc)
`- mbed-os (6e0d01cd13e8)

chrissnow, 2017-10-13:
@toyowata could you try with the ARM compiler?
Having just tested GCC_ARM myself it does seem to work.

chrissnow, 2017-10-13:
I think the cause may be here
https://github.com/ARMmbed/mbed-os/blob/df88a9dcc2700535ca00bdf8dbeb86ab048c396f/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_NXP/lpc17_emac.c#L149-L153
It looks like It's missing the ARM equivalent

Changing that to include ARM fixes it.

0xc0170, 2017-10-13:
That line has been there for 2 years, interesting

chrissnow, 2017-10-13:
It also explains why I have next to no RAM when using Ethernet...

chrissnow, 2017-10-13:
I can sort out a PR if this is the true cause, It certainly fixes a number of issues for me.
Let me know what you think.

geky, 2017-10-13:
This looks like the right fix, although I don't know what could have caused it to break in 5.6, @kjbracey-arm thoughts?

@chrissnow If you're interested in creating a pr that would be great

chrissnow, 2017-10-15:
PR #5322 has the fix.

kjbracey-arm, 2017-10-16:
No particular knowledge of memory layout for this board - I think maybe @mikaleppanen may have looked at this while updating to lwIP 2.0

mikaleppanen, 2017-10-16:
Maybe updated lwip stack uses a bit more memory. Haven't tested the lwip 2.0 with this board, but there is free memory in the AHBSRAM1, so the area can be located there. Change is ok for me.

PR #5322 (2017-11-09): LPC1768: Fix ETHMEM_SECTION placement for ARM

## Description
Ethernet on LPC1768 is broken in 5.6 as per #5298
This corrects the placement of Ethernet memory with the ARM compiler.

## Status
**READY**

## Migrations
NO

## Related PRs

branch | PR
------ | ------

## Todos

## Deploy notes

## Steps to test or reproduce
Build the TCPSocket_Example with 5.5.7 and it will work, any 5.6\master will fail to get an IP, using static IP will still fail to work properly.

Fixes #5298
closed 2017-11-09 by 0xc0170