ARMmbed/mbed-os #5301: Project exporters and CMSIS Pack for NCS36510 target

danclement, 2017-10-11:
**Need advice for how best to implement a minimal viable CMSIS pack for target NCS36510, how to test it, and how to integrate the pack into mbed so the project exporter can support Keil (and GCC and IAR if they can use the CMSIS pack also).**

@theotherjimmy @maclobdell @jacobjohnson-ON 

### Summary of question points

1. What is included in a minimal CMSIS device pack?
2. How do we test this CMSIS pack?
3. How do we bring it into mbed?
4. How do we enable support in the exporter to use this cmsis pack, for all toolchains including Keil?

**Target**
NCS36510

**Toolchain:**
GCC_ARM|ARM|IAR

JonatanAntoni, 2017-10-12:
Hi @danclement,

Please take a look into [CMSIS-Pack documentation](http://www.keil.com/pack/doc/CMSIS/Pack/html/index.html) to make yourself familiar with it. A Pack is simply an archive containing a pack description (pdsc). What you need to put into this archive depends on your usage intend. For a device family pack (DFP) we typically add:

- device description
- startup code (as a component)
- driver implementations (i.e. HAL library)
- blinky examples

Please feel free to let us know about your CMSIS Pack related questions and feedback on the [CMSIS GitHub site](https://github.com/ARM-software/CMSIS_5/issues).

Best,
Jonatan

danclement, 2017-10-12:
@JonatanAntoni, thanks for your link to the CMSIS GitHub site.  I'm familiar with the documentation and I have been studying it.

Unfortunately my points must not have been clear, I'm seeking advice for how best to implement and integrate a CMSIS pack for use inside mbed, and how to attach it to the project exporter tools.

For general CMSIS pack questions I will use the general CMSIS GitHub issues list, but my questions remain with regards to mbed.

Thanks,
Dan

theotherjimmy, 2017-10-12:
Hi @danclement,

GCC does not use CMSIS-PACKs.

The Keil exporter looks in the arm-pack-manager's cache for it's information. We need to add the NCS36510 to the cache, with [`pack_manager.py`](https://github.com/ARMmbed/mbed-os/blob/master/tools/arm_pack_manager/pack_manager.py#L79-L112) and add the `device_name`, which corresponds to a device in the DFP pack, into [`targets.json`](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3191-L3227). Then we can start testing the Keil exporter (in CI!).

IAR requires all of the same steps as Keil, and an additional step: we need some information to fill out the template with. This info is in [`iar_definitions.json`](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/iar_definitions.json). After updating that, we can also test the IAR exporter in CI.

theotherjimmy, 2017-10-12:
Further, Mbed only uses the device description from the CMSIS-PACK at the moment.

danclement, 2017-10-12:
Thanks @theotherjimmy, I'll work on getting the CMSIS pack for NCS36510 put together.  When you mention the device description, is this the SVD file or just the description in the pdsc file?

Should I prepare the pack and then send it you, or am I expected to make the changes above and do pull requests?

Thanks

theotherjimmy, 2017-10-12:
@danclement Good catch! I was referring to the device description in the `pdsc` file.

danclement, 2017-10-12:
P.S. I sent my prototype pack to Mac yesterday, maybe he can help us figure out how to hack it into mbed and if what I made is good enough for now. @maclobdell

theotherjimmy, 2017-10-12:
Nice! I'll take a look.

closed 2019-10-31 by ciarmcom