ARMmbed/mbed-os #5307: NCS36510: Inconsistent Ticker callback calls when scheduled many tickers at the same time

maciejbocianski, 2017-10-12:
Inconsistent Ticker callback calls when scheduled many tickers at the same time

### Description
- Type: Bug
- Related PR: https://github.com/ARMmbed/mbed-os/pull/5261
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
master

**Expected behavior**
Ticker callback is executed exactly 16 times

**Actual behavior**
Ticker callback is executed 2 or 32 times

**Steps to reproduce**

Run following code

```
#define MULTI_TICKER_TIME_MS 100
#define TICKER_COUNT 16
Ticker ticker[TICKER_COUNT];
uint32_t multi_counter;

void increment_multi_counter(void)
{
   core_util_atomic_incr_u32(&multi_counter, 1);
}

const uint32_t extra_wait = 5; // extra 5ms wait time
multi_counter = 0;
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].attach_us(callback(increment_multi_counter), MULTI_TICKER_TIME_MS * 1000);
}

Thread::wait(MULTI_TICKER_TIME_MS + extra_wait);
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].detach();
 }

TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter);
```

0xc0170, 2017-10-12:
cc @danclement  @jacobjohnson-ON @maclobdell

0xc0170, 2017-10-12:
Same behavior (same issue) is also for KL46Z cc @mmahadevan108

danclement, 2017-10-12:
Our us ticker on NCS36510 is a software implementation that wraps two 16 bit counters and makes them look like one.  Inside this software wrapper are interrupts and ISRs.  I suspect that these nested interrupt handlers are causing priority issues in the overall system as the ticker and what is attached to it are basically an interrupt and ISR also.

Maybe our us ticker needs to disable other interrupts to make sure it works?  Any advice here Mac?

0xc0170, 2017-10-13:
The ticker API supports 16 bit tickers (there are few more changes needed as I recall (related to the maximum delta that is predefined now?), @pan- can confirm).

maciejbocianski, 2017-10-23:
ping

bulislaw, 2017-10-27:
@danclement @jacobjohnson-ON @maclobdell @0xc0170 ping

0xc0170, 2017-10-27:
I am looking at this, I noticed one issue with fire interrupt, I'll send a fix.

@maciejbocianski Can you retest with these changes

`fire_interrupt` should be
```

void us_ticker_fire_interrupt(void)
{
    us_ticker_target = 0;
    NVIC_SetPendingIRQ(Tim1_IRQn);
}
```

This might help, then we shall check as mentioned above, two isr routines.

maciejbocianski, 2017-10-30:
@0xc0170 
I would very much like to do that, but dont have NCS36510 device,

maciejbocianski, 2017-11-21:
@0xc0170 maybe we should clone this issue for KL46Z?

0xc0170, 2017-11-21:
> @0xc0170 maybe we should clone this issue for KL46Z?

If there is any, yes please do

maciejbocianski, 2017-11-23:
issue was cloned for KL46Z see https://github.com/ARMmbed/mbed-os/issues/5564

closed 2017-11-23 by maciejbocianski