ARMmbed/mbed-os #5308: RTC set/get time issue on NCS36510

mprse, 2017-10-12:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
47c890da8

**Expected behavior**
When RTC time value is set and read in sequence, then set value should be returned by the read function.
After setting RTC time we should have about 1 second to finish read operation. It is long enough to assume that the set value should be read.

**Actual behavior**
On NCS36510 board if we perform time set and get operations in sequence, then it looks like about 30 sec have elapsed between set and get operations.

**Steps to reproduce**
```
set_time(1256729737);
seconds = time(NULL);
TEST_ASSERT_EQUAL(1256729737, seconds);  // FAIL: Expected 1256729737 Was 1256729797
```

0xc0170, 2017-10-12:
cc @danclement @jacobjohnson-ON @maclobdell

danclement, 2017-10-12:
FYI, there is no posted documentation that I could find that would make it clear the RTC is associated with this set_time and time functions...  Can you link me to something?

I have no idea why this wouldn't work.  We will have to look into it.

mprse, 2017-10-13:
@danclement 
Hi, the source code of set_time() and time() functions indicates that time is based on RTC:
https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_rtc_time.h
https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_rtc_time.cpp

0xc0170, 2017-10-19:
@danclement Any update?

danclement, 2017-10-19:
Sorry, no updates yet.  Resource and priority issues.  Same for the other issues.

We talked to @maclobdell about this and we discussed that we need some time to resolve, likely several weeks.

cmonr, 2018-01-11:
@mprse @danclement Has any progress been made on this?

mprse, 2018-01-12:
> @mprse @danclement Has any progress been made on this?

No progress. Maybe in the near feature I will have NCS36510 board, then I'll try to investigate and fix the problem.

bulislaw, 2018-01-15:
That is open for 3 months now, to get that merge we should remove the RTC capabilities of this device. @maclobdell what are your thoughts?

danclement, 2018-01-17:
As I informed Mac a while back we have put our mbed project on hold for now.  We will try to pick it up again in Q218.

cmonr, 2018-01-17:
@danclement Thanks for the update. For the time being, I'll close this issue.

Feel free to reopen it or create a new issue when the time is appropriate.

closed 2018-01-17 by cmonr

0xc0170, 2018-01-18:
@mprse Until this is resolved, we shall disable RTC for this target (separate commit that references this issue, so we can revert it once the fix is in place).

mprse, 2018-05-09:
Can we re-enable this issue?

I have provided fix here: PR# https://github.com/ARMmbed/mbed-os/pull/6852.

bulislaw, 2018-05-09:
GitHub linked them so I don't think there's point in reopening it, lets just make sure the fix gets in :)

mprse, 2018-05-11:
PR# #6852 has been merged. RTC support has been re-enabled for NCS36510.
Issue has been fixed.