ARMmbed/mbed-os #5309: MAX32630 MAX32625:  Program hangs when scheduled many tickers one after the other with the same time interval

maciejbocianski, 2017-10-12:
When schedule many Tickers one after the other with the same time interval 100ms and then wait 105ms it's expected that after this time all Tickers will be fired once. But we get other results. There is no action test hangs and timeout occurs

Problem noticed in https://github.com/ARMmbed/mbed-os/pull/5261 in the following test
```
/** Test many tickers run one after the other

    Given many Tickers
    When schedule them one after the other with the same time intervals
    Then tickers properly execute callbacks
    When schedule them one after the other with the different time intervals
    Then tickers properly execute callbacks
 */
void test_multi_ticker(void)

```
Sample code for reproduction
```
#define MULTI_TICKER_TIME_MS 100
#define TICKER_COUNT 16
Ticker ticker[TICKER_COUNT];
uint32_t multi_counter;

void increment_multi_counter(void)
{
   core_util_atomic_incr_u32(&multi_counter, 1);
}

const uint32_t extra_wait = 5; // extra 5ms wait time
multi_counter = 0;
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].attach_us(callback(increment_multi_counter), MULTI_TICKER_TIME_MS * 1000);
}

Thread::wait(MULTI_TICKER_TIME_MS + extra_wait);
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].detach();
 }

TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter);
```

### Description
- Type: Bug
- Related PR: https://github.com/ARMmbed/mbed-os/pull/5261
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
MAX32630, MAX32625

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
master

**Expected behavior**
Ticker callback is executed exactly 16 times

**Actual behavior**
Test hung and timeout occurs

0xc0170, 2017-10-13:
cc @kgills @jeremybrodt

maclobdell, 2017-10-13:
@ARMmbed/team-maximintegrated

maciejbocianski, 2017-10-23:
ping

jeremybrodt, 2017-10-25:
@jessexm

bulislaw, 2017-11-03:
ping @ARMmbed/team-maximintegrated @jessexm

jessexm, 2017-11-10:
Currently testing fix.

0xc0170, 2017-11-16:
@jessexm Let us know if any problem

maciejbocianski, 2018-01-09:
@jessexm what is the status of the fix?

0xc0170, 2018-06-22:
Two fixes addressed this issue (referenced above)

closed 2018-06-22 by 0xc0170