ARMmbed/mbed-os #5310: Missing content-format for static resources

pawel-biczysko-tridonic-com, 2017-10-12:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STMF32F0

**Toolchain:**
All

**mbed-os sha:**
https://github.com/ARMmbed/mbed-os/commit/47c890da8370e5e926fcc4d350c4b1b78f4f09fb & https://github.com/ARMmbed/mbed-os/commit/8828635da469162cf2854b5287561c663fb96e72

**Expected behavior**
Content format "text/plain" should be included in the CoAP response to the static resource get request as it is for the dynamic resources. 

**Actual behavior**
If the selected content-format is "text/plain" (the default one) it is not included in the CoAP response message.

**Probable cause**
CoAP requests to static resources are handled by the sn_grs_process_coap function in sn_grs2.c and along the way there is this check for the not set content-type which excludes the possibility to set "text/plain" for static resources:

https://github.com/ARMmbed/mbed-os/blob/5faf4b26c5954d15c7c1cccac6498e0c690ad101/features/FEATURE_COMMON_PAL/mbed-client-c/source/libNsdl/src/sn_grs2.c#L706-L714

@karsev @MarceloSalazar

pawel-biczysko-tridonic-com, 2018-03-14:
This is the quick solution that I have been using. It removes check for 0 content type allowing to select "text/plain"

```
if (status == COAP_MSG_CODE_RESPONSE_CONTENT) {
/* Add content type if other than default */
    if (resource_temp_ptr->static_resource_parameters) {
        response_message_hdr_ptr->content_format = (sn_coap_content_format_e) resource_temp_ptr->coap_content_type;
}
```

MarceloSalazar, 2018-04-19:
Verified with the latest cloud client (1.3.1) and content format text/plain is now part of the response.
https://github.com/ARMmbed/mbed-cloud-client/blob/1.3.1/mbed-client/mbed-client-c/source/sn_grs.c

Closing - please re-open if needed.

closed 2018-04-19 by MarceloSalazar