ARMmbed/mbed-os #5333: Undefined variable rtos::Thread::wait((uint32_t) ms)

FusedShadowX, 2017-10-17:
### Description
- Type: Bug | Question
- Priority: Blocker | Major

---------------------------------------------------------------
## Bug
This bug occurs on Platformio on the linking of firmware.elf. The error is thrown inside mbed_wait_api_rtos on declaration of void wait(it us). For some reason it cannot find the static declaration of Thread::wait although the framework has not been changed and it has been removed and installed twice. 

**Target**
NPX mbed LPC1768 | Freescale Kinetic K66F

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
mbed_os platformio version. 

**Expected behavior**
Compilation should not stop as the declaration and call are correct. 

**Actual behavior**
Throws the "Undefined variable rtos::Thread::wait" and stops build. 

**Steps to reproduce**
It may be connected with the calls to pwm base functions (pwmout_init, pwmout_write) or calls to the core_util_critical_section_enter() and exit(). 
----------------------------------------------------------------

## Question
Is this issue a mbed_os problem or PlatformIO's? 
I'm sorry for not giving an example as I'm not connected from a computer. Furthermore, there is not any example to be given as none of my main code calls any of the wait functions.

pan-, 2017-10-17:
Did you define the flag needed to include the this into your build ? See [here](http://docs.platformio.org/en/latest/frameworks/mbed.html#configuration)

FusedShadowX, 2017-10-17:
Sorry I didn't mention it. Yes I have used both the flags for events and rtos. I also removed them, build the project successfully, clean it and re-enable the flags, but nothing changed.

0xc0170, 2017-10-18:
@FusedShadowX How to reproduce this issue? What version of mbed OS ?

based on the details you shared above, the issue might be somewhere else than in this codebase (mbed-os code).

FusedShadowX, 2017-10-18:
The reproduction of the problem is unknown. This issue was thrown on the build without a major change, I just added a header file with some functions and build it. Previously, everything was ok. The mbed version as described in the package.json of PlatformIO in the framework_mbed folder is 4.152. 

I have my computer also, so i can give you the thrown error:

> c:/users/fusedshadowx/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m/fp
v4-sp/softfp/crt0.o: In function `_start':
(.text+0x4a): undefined reference to `__wrap_main'
.pioenvs/frdm_k64f/FrameworkMbedCore/platform/mbed_wait_api_rtos.o: In function `wait_us':
mbed_wait_api_rtos.cpp:(.text.wait_us+0x2a): undefined reference to `rtos::Thread::wait(unsigned long)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\frdm_k64f\firmware.elf] Error 1

So I made a mistake on the title of this error, _"Undefined reference"_
Above this is the compilation of the other files, nothing more.

0xc0170, 2017-10-18:
Isn't this an error :undefined reference to__wrap_main ?

Do you have `-Wl,--wrap,main` in the ld flags?

pan-, 2017-10-18:
@FusedShadowX Mbed OS 4.152 does not exist. If before the PlatformIO update everything was OK, I suggest you report the issue to PlatformIO.

FusedShadowX, 2017-10-18:
Yes I have the ld flags. As @pan- said it must be PlatformIO's problem. I'll report it to them. Thanks everyone for your instant help.

closed 2017-10-22 by FusedShadowX