ARMmbed/mbed-os #5341: FATFS: mkfs

chrissnow, 2017-10-18:
### Description
- Type: Bug
- Related issue:
- Priority: Blocker Minor

---------------------------------------------------------------
## Bug
If you have a device that has 4096 512/528b sectors and try to format without specifying the AU it fails.
https://github.com/ARMmbed/mbed-os/blob/20d93bf78d5172b6a0f1c1065058d641e4e0f2e1/features/filesystem/fat/ChaN/ff.cpp#L4172-L4174
Which results in FAT16 as the result since nclst = 0x1000
Yet this then fails here as there is not enough sectors for FAT16
https://github.com/ARMmbed/mbed-os/blob/20d93bf78d5172b6a0f1c1065058d641e4e0f2e1/features/filesystem/fat/ChaN/ff.cpp#L4204-L4207
and returns

```
FR_MKFS_ABORTED
```
If you format with an AU of 2
```c++
FATFileSystem::format(&flash,2)
```
It will succeed with FAT12

**Expected behavior**
Format's the device as FAT12
**Actual behavior**
```
FR_MKFS_ABORTED
```
**Steps to reproduce**
Use a 16mbit DataFlash which has 4096 sectors of 512/528 (or anything similar)
```c++
FATFileSystem::format(&flash); //this fails
FATFileSystem::format(&flash,2);//this works
```

Further info. https://github.com/ARMmbed/dataflash-driver/issues/3
CC @geky @marcuschangarm

geky, 2017-10-20:
Thanks for raising this!

internal prioritization [IOTMORF-1634](https://jira.arm.com/browse/IOTMORF-1634)

yossi2le, 2018-06-19:
@chrissnow @cmonr @0xc0170  @geky 
Hi All 
This bug has been opened October 18 2017 and on November 10 2017 the ChanFs was upgraded to R0.13a.  I have tried to reproduce the bug but it is not reproducible any more. 
So can someone close this bug please?

Thanks

yossi2le, 2018-06-19:
@cmonr  10x