ARMmbed/mbed-os #5349: I2C driver port missing a command, can get stuck in loop forever

danclement, 2017-10-19:
### Description
From if section below there is no operator that increments the write variable causing the system to get stuck in while loop forever.

Reviewing the code it's apparent that a line of code was missed from the HAL port.

The fix is to add write++ at the end of the if section.  This needs to be implemented and checked in mbed but in our other SDK it exists and is tested so the risk is low.

I listed this as minor because it's very easy to make the change locally and is low risk.  We will update the code and do the mbed tests and formally make a pull request as time and resources permit.

- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
After bitwise write of data write variable should increment and the while loop should eventually exit as expected.

**Actual behavior**
For the I2C_CMD_RDAT8 command, the code gets stuck in the while loop because write variable never increments.

**Steps to reproduce**
Use the I2C and send the I2C_CMD_RDAT8 command, code will get stuck in while loop.

From file: ncs36510_i2c.c

```
       while (write < len) {

       /* Send write command */
        SEND_COMMAND(I2C_CMD_WDAT8);

        if(buf[write] == I2C_CMD_RDAT8) {

            /* SW work around to counter FSM issue. If the only command in the CMD FIFO is the WDAT8 command (data of 0x13)
            then as the command is read out (i.e. the FIFO goes empty), the WDAT8 command will be misinterpreted as a
            RDAT8 command by the data FSM; resulting in an I2C bus error (NACK instead of an ACK). */

            /* Send 0x13 bit wise */
            SEND_COMMAND(I2C_CMD_WDAT0);
 
            SEND_COMMAND(I2C_CMD_WDAT0);
 
            SEND_COMMAND(I2C_CMD_WDAT0);
 
            SEND_COMMAND(I2C_CMD_WDAT1);
 
            SEND_COMMAND(I2C_CMD_WDAT0);
 
            SEND_COMMAND(I2C_CMD_WDAT0);
 
            SEND_COMMAND(I2C_CMD_WDAT1);
 
            SEND_COMMAND(I2C_CMD_WDAT1);

            // Add write++ command here
           write++

        } else {
            /* Send data */
            SEND_COMMAND(buf[write++]);
        }
        SEND_COMMAND(I2C_CMD_VRFY_ACK);
```

maclobdell, 2018-08-16:
@danclement - did this issue get resolved?  It seems that the fix was understood, right?  Did a pull request get made to patch the issue?  If so, please provide the reference.

danclement, 2018-08-26:
Mac,

I thought we formalized this fix and had it merged.  I’ll have to look into this for the history.  It will take some time.

Best Regards,
Dan Clement

deepikabhavnani, 2018-12-28:
As per the conversation above, this commit should have resolved the issue: https://github.com/ARMmbed/mbed-os/commit/36d335baf4eace9513181636db1714666b621dfb#diff-dc408e98fca5637b29cbf2c7e9ceacfe

closed 2018-12-28 by deepikabhavnani