ARMmbed/mbed-os #5350: K64F: SPI dummy data not working as intended

kjbracey-arm, 2017-10-20:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F (and other Freescale?)

**mbed-os sha:**
6e0d01cd13

**Expected behavior**
When RX len > TX len in SPI block write,  the excess TX bytes should be the fill value.

**Actual behavior**
It actually reads off the end of the TX buffer and transmits those bytes.

The core implementation of DSPI_MasterTransferBlocking looks encouraging, with separate remainingSendByteCount and remainingReceiveByteCount, but both of those are always set to the same value, the maximum of RX and TX len.

Bug observed while inspecting SPI traffic to the Atmel RF233 radio. Doesn't affect me - the readio doesn't care about the extra data, but apparently SD cards do? And obviously running off the end of buffers is bad.

kjbracey-arm, 2017-10-20:
Actually, this might be worse - I don't see why it wouldn't also write off the end of the RX buffer when TX len > RX len. 

The problem is partially concealed by pointer NULL checks, though, so pure reads and writes with the other pointer being NULL seem to do the right thing - the bug only occurs when both pointers are non-NULL and lengths differ.

0xc0170, 2017-10-27:
cc @mmahadevan108

maclobdell, 2018-12-11:
@mmahadevan108  Please share an update.

mmahadevan108, 2018-12-11:
I logged a ticket in our JIRA system to fix the DSPI driver 
_NXP JIRA reference https://jira.sw.nxp.com/browse/KPSDK-24834_

maclobdell, 2019-01-14:
@mmahadevan108 - any update on this issue?  schedule for getting a fix?

mmahadevan108, 2019-01-15:
This is what I got from our internal team
____________________________
The DSPI transfer is duplex, so the transfer size is the same for sending and receiving. 
_____________________________

maclobdell, 2019-02-05:
@kjbracey-arm  & @mmahadevan108  It seems there is still some confusion about this issue.  Can you clarify more exactly what the issue is, specifically, where a fix might be needed.

maclobdell, 2019-06-17:
closing due to inactivity.  please re-open if necessary.

closed 2019-06-17 by maclobdell